spacetelescope/hstcal #278: Improve command line usage text for ACS executables

mdlpstsci, 2017-11-16:
While the ACS executables have existing usage messages, they are incomplete in that the options have no explanation (i.e., what does "-t" mean?).

pllim, 2017-11-16:
If it is the same as `calacs.e`, `-t` means include timestamp and `-v` means verbose.

pllim, 2017-11-16:
http://acstools.readthedocs.io/en/latest/acsrej.html is meant as an user-facing documentation of HSTCAL usage because there is no (easy) way to build Sphinx doc with pure C code, so I stuck the doc stuff in ACSTOOLS. You are welcome to move the doc somewhere more obvious if you wish.

While you are at it, there should be a `.e` suffix for executable.

p.s. On hindsight, option that uses a word like `-shadcorr` should have been `--shadcorr` to be more consistent with common usage out there but that boat has sailed.