spacetelescope/hstcal #280: Fix regression from #275

jamienoss, 2017-11-16:
Woops, looks like I broke STIS.

The following test segfaulted both on `nott` (linux) and `pembry` (OSX):
On `nott` [stis/wavecal/wavecal_lev2](https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&qid=7398)
On `pembry` [stis/wavecal/wavecal_lev2](https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&qid=7399)

I ran this through the regression suite but with my own code which I've just noticed had all the STIS tests turned off, [here](https://github.com/jamienoss/regress/blob/master/regress.py#L453-L457).
```python
...
 # Start with ACS (have to start somewhere)
        exe = os.path.join(execPath, 'calacs.e')
        checkExeExists(exe)
        for test in acsInput:
            testItem = TestQueueItem(test, exe, outPath)
            testQueue.put(testItem)

        # Then STIS
        # cmd = os.path.join(execPath, 'calstis.e')
        # for test in stisInput:
        #    testItem = TestQueueItem(test, cmd, outPath)
        #    testQueue.put(testItem)

        # Then WFC3
        exe = os.path.join(execPath, 'calwf3.e')
        checkExeExists(exe)
        for test in wf3Input:
            testItem = TestQueueItem(test, exe, outPath)
            testQueue.put(testItem)
...
```

jamienoss, 2017-11-16:
@mdlpstsci FYI

pllim, 2017-11-16:
I saw two STIS failures:

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=122812907

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=122813017

jamienoss, 2017-11-27:
Looks like this may still be a problem https://ssb.stsci.edu/pandokia/pandokia.cgi?query=day_report.2&host=%2A&project=%2A&test_run=conda-conda_tests_py3_2017-11-26-10%3A30%3A02

jamienoss, 2017-11-27:
Hehe, woops, major schoolboy:
[source](https://github.com/spacetelescope/hstcal/blob/e07f1c99d485fab135690555effdbbcfb0413367/pkg/stis/calstis/cs6/cs6.c#L181-L188)
```c
...
if (MkOutName (finput, isuffix, osuffix, nsuffix,
                foutput, STIS_FNAME))
	        freeOnExit(&ptrReg);
	        exit (ERROR_RETURN);
...
```
Unguarded `if`.

There are a few of these within #275

jamienoss, 2017-11-27:
I normally do a diff of the build output of master vs PR so as to check for introduced compiler warnings. I clearly must not have in this case otherwise I would have found these.

PR #282 (2017-11-21): Remove incorrect non-ptr registrations
Fixes #280

Regression introduced from #275
commit 88232616 (2017-11-21): Remove incorrect non-ptr registrations

Fixes #280

Regression introduced from #275
PR #283 (2017-11-27): Fix unguarded IFs introduced in #275
Fixes #280

This also addresses the "-Wmisleading-indentation" compiler warnings generated
from the following files, also from #275

pkg/stis/calstis/lib/reffiles.c
pkg/stis/calstis/cs12/cs12.c
pkg/acs/calacs/lib/reffiles.c
pkg/wfc3/calwf3/lib/reffiles.c
commit c795f13b (2017-11-27): Fix unguarded IFs introduced in #275

Fixes #280

This also addresses the "-Wmisleading-indentation" compiler warnings generated
from the following files, also from #275

pkg/stis/calstis/lib/reffiles.c
pkg/stis/calstis/cs12/cs12.c
pkg/acs/calacs/lib/reffiles.c
pkg/wfc3/calwf3/lib/reffiles.c
closed 2017-11-27 by jamienoss