spacetelescope/hstcal #281: Calwf3 on-the-ramp fitting of variable IR data should issue a warning/flag.

npirzkal, 2017-11-16:
WFC3 IR observations can be affected by a large amount of HeI emission from the bright Earth Limb. The amount of HeI varies during he course of a visit (usually most at the beginning or end of a visit). On-the-ramp fitting assumes and requires that the underlaying signal have a constant rate in e/s (aside from cosmic ray hits), which is violated in this case. We have found that the result are images where the photometry is strongly affected and in spectroscopic/Grism observations the results are images that cannot be properly background subtracted (WFC3 isr 2017-05 by Pirzkal et Al)
calwf3 should report of the quality of the on-the-ramp fitting step and issue a warning and possibly a flag in the header of files when this is the case. This will allow users to be made aware that the data were negatively affected by a variable background and that they should take the appropriate steps. This is something not currently possible and it is difficult to visually assess if on the ramp negatively affected the data.

mdlpstsci, 2022-01-12:
There is a project for 2022 Q3 - Time Variable Background in WFC3 IR.  Much discussion has been done on the wfc3-flb slack channel.  Work done for this project may make this ticket obsolete.

mdlpstsci, 2023-09-05:
As of this date, September 2023, the time variable IR background for WFC3 IR is still a project for which I have not yet received thorough information.

closed 2025-10-15 by pllim