spacetelescope/hstcal #287: Modify the usage of the crmask parameter for ACSREJ standalone component of CALACS

mdlpstsci, 2018-01-05:
Modify the usage of the crmask parameter of the ACSREJ standalone component of CALACS so the user can turn flagging of cosmic rays in the input images ON or OFF without having to resort to having to copy the CRREJTAB to a temporary file, updating items in the table, and then providing this temporary table to the ACSREJ task.  The current crmask command line parameter (datatype bool) does not actually act like a bool in that it does not override the flagging functionality.   If the user tries to turn the flagging OFF, the program will then consult the CRREJTAB for further action.  This issue was submitted at the request of ACS Instrument Team member.

pllim, 2018-01-09:
I am guessing it was @tddesjardins? :smile: (c/c him here so he gets notified.)

This proposal sounds reasonable. However, it is important to keep some kind of history (could be as simple as adding `HISTORY` to primary header) that things were overridden via command line and the `CRREJTAB` file stated in output header no longer applies. This is important in case someone inherited the output file and wants to reproduce the exact result.

mdlpstsci, 2023-09-05:
This is a duplicate to Issue https://github.com/spacetelescope/hstcal/issues/572 which the topic is described more thoroughly. Closing this issue and using https://github.com/spacetelescope/hstcal/issues/572 going forward.

closed 2023-09-05 by mdlpstsci