spacetelescope/hstcal #298: CALSTIS failure

pllim, 2018-01-26:
Truth appears to be missing:

https://ssb.stsci.edu/pandokia/pandokia.cgi?query=detail&key_id=134863803

stscirij, 2018-01-26:
No it looks like a fail in the run

ERROR    Output file `out/TEMPX1D_calstis_lev1_FUVspec_x1d.fits' already exists.
calstis0 failed.

stscirij, 2018-01-26:
I removed the TEMP files from the out/ directory, the test should run OK now

jamienoss, 2018-01-26:
It may be due to all the recent killing and restarting of tests for 2018.1. A similar situation was found with a previous WFC3 test. There is no easy bulk cleanup procedure for the suite (that I am aware of). We could look at the test in question and see if this file is in the test's pre-run rm list. If it is not and pandokia was killed before this test completed then this temp file would not be removed. I am assuming that this is a tmp file due to the 'TEMP" prefix.

stscirij, 2018-01-26:
It's not in the pre-exec steps.  We should definitely put in some lines to remove the TEMP files that could be created.  I'll do that.

pllim, 2018-01-26:
Thanks for the quick fix!

closed 2018-01-26 by pllim