spacetelescope/hstcal #311: Reed-Solomon Decoding Errors

tddesjardins, 2018-05-25:
There is currently no check for Reed-Solomon (RS) decoding errors in `calacs`. RS errors are a result of compression in ACS data and are listed in ACS documentation as being marked in the DQ array with bit value 1. They appear as long, thin line segments similar to cosmic rays, with all of the segments parallel to each other and an inflection at the amplifier boundary. I've attached a screenshot from dataset J94TPKMIQ showing an extreme example of this effect.

[image]

At the moment, someone has marked these in the DQ arrays of raw files in the archive, but nobody remembers who did it or how, and the headers of the raws were not updated with that information. Further, they were marked with the wrong DQ flag, so they appear as stable hot pixels at the moment, then subsequently get marked as cosmic rays by `acsrej` and `AstroDrizzle`. If the raws are ever generated again from pod files, we will lose this flagging altogether, which is the biggest concern.

The ACS team would like to build in automatic detection and flagging with DQ value 1 of these features as part of `calacs`. To our knowledge, this effect is limited to datasets that used compression. I'm not sure if there's a simple boolean keyword in the headers for that, but there is information about the compression in the first SPT header extension (looks like BUFSIZE = 0 is uncompressed and anything non-zero is compressed), so we can use that as a starting point to check if RS error flagging needs to be performed at all.

The ACS Team would like to include this as part of HSTDP 2018.3, however it is low priority and can wait until 2019.1 if necessary.

tddesjardins, 2018-05-25:
Also as an addendum, it would be incorrect to flag these as cosmic rays in addition to RS errors, so some logic in `acsrej` to not mark pixels in the DQ array that already contain a 1 with the cosmic ray flag would be a good idea if doable. As for `AstroDrizzle`, I'm not sure if we can make that happen, but I'll mention it to Warren Hack.

pllim, 2018-05-25:
@stsci-hack , do you remember any history regarding to Reed-Solomon DQ flagging? Please advise.

pllim, 2018-06-14:
Update: Tyler found out from Lisa Sherbert that this might be populated during Generic Conversion before data hits CALACS. Thus, if this can be fixed upstream, no action is needed for CALACS, where SOFTERR is defined correctly in `include/acsdq.h` but not used anywhere.

@tddesjardins , please link any appropriate JIRA ticket here. Thanks!

tddesjardins, 2018-06-14:
There is a discussion of flagging RS decoding errors and missing data in PR 69098 in the STScI Problem Report system. According to Lisa, the RS flagging DOES happen during generic conversion, so now we'll move over to JIRA to figure out why things are being flagged as hot pixels.

closed 2018-06-14 by tddesjardins

tddesjardins, 2018-06-15:
[Here](https://jira.stsci.edu/browse/HSTSDP-216) is a link to the JIRA issue for this in the HSTDP group.