spacetelescope/hstcal #316: cfitsio updates

rendinam, 2018-06-20:
This issue is meant to track `cfitsio` library updates that become available.

`hstcal` now uses the `cfitsio` package provided on the Astroconda channel. When `hstcal` needs to use a newer version than what Astroconda provides, this issue will be used to indicate that intention.

Updates to `cfitsio` will be provided at a time determined by the following rules:
  1) Security update (whether or not it affects functionality used by `hstcal`): Yes, update the library.
  2) API Change, backward compatible: Low priority; update when possible/convenient.
  3) API Change, breaks compatibility: Yes, update the library.

---

`cfitsio` updates:

Version 3.45 - May 2018
                   
  - New support for reading and writing unsigned long long datatypes.
    This includes 'implicit datatype conversion' between the unsigned long
    long datatype and all the other datatypes.
    
  - Increased the hardcoded NMAXFILES setting for maximum number of
    open files from 1000 to 10000.
                   
  - Bug fix to fits_calc_binning wrapper function, which wasn't filling
    in the returned float variables.
    
  - Fixed a parsing bug for image subsection and column binning range
    specifiers that was introduced in v3.44.

jamienoss, 2018-08-03:
@rendinam @jhunkeler (at some point) can we get 3.450 built and distributed via astroconda, please?

Also, what would the issues be with building these all with ``-g`` so that the debug symbols are added? If not wanted for the release versions under ``astroconda`` we could mirror them all on``astroconda-dev`` that included the symbols? Any thoughts...

pllim, 2018-08-21:
Now that `cfitsio` is an external dependency, can we close?

mdlpstsci, 2022-01-11:
HSTCAL receives notifications when updated versions of `cfitsio` are released, and as noted, it is now an external dependency.  Action should be taken as noted above for security or an API change which breaks backward compatibility.

closed 2022-01-11 by mdlpstsci