spacetelescope/hstcal #318: Update documentation

stscicrawford, 2018-06-20:
The current documentation points to the STScI website and does not look to be currently up to date.   It would be useful to migrate the documentation to be part of the repository with more descriptive information about the package.

pllim, 2018-06-20:
@stscicrawford , can you please provide the link to the outdated doc mentioned above?

HSTCAL doc is tricky because there is no precedence in the branch to build Python Sphinx doc in a pure-C package. I buried CALACS usage doc in http://acstools.readthedocs.io/en/stable/ . That said, HST users mostly refer to Data Handbook for the appropriate instrument for such info, e.g., http://www.stsci.edu/hst/acs/documents/handbooks/currentDHB/acs_Ch31.html . Handbook update happens as needed (or per cycle) and it is an effort on the INS side.

stscicrawford, 2018-06-20:
Well, this wold be the page I'm referring to:
http://www.stsci.edu/institute/software_hardware/hstcal

I'd be happy with the tools packages and the handbooks holding the information, but we should point to that from the README here at the very least and make sure the link on the top is pointing to page with more useful information.

pllim, 2018-06-20:
@SaOgaz , any comment on this?

SaOgaz, 2018-06-22:
Let me think about this and get back to you.

SaOgaz, 2018-06-29:
Okay, sorry for the delay.  I agree with both @pllim and @stscicrawford.  That being said, a lot of the tools packages documentation is VERY sparse and needs filling out.  I will hopefully be working with the STIS team in the future to add more narrative style hstcal related documentation, but someone needs to do this for wfc3 as well. And maybe calcos (not sure what the state of their documentation is at the moment)?  @hcferguson did tag me as sort of lead on documentation, so I can follow up on this in more detail offline and see about getting things moving forward.

SaOgaz, 2018-06-29:
If everyone is satisfied with that, I can close this issue and open new issues on the respective tool repos.

mdlpstsci, 2018-06-29:
I would say there should be some standard or template, where possible, for the documentation.  For some of the tools, the parameter descriptions should include: name, useful description, datatype, required or optional, and default values (or a description the value is computed on-the-fly).  I can see this is lacking in places.

SaOgaz, 2018-07-02:
Yeah, @mdlpstsci.  Most of our repos use the Numpy docstring standard for this: https://numpydoc.readthedocs.io/en/latest/format.html#docstring-standard

SaOgaz, 2018-07-02:
Actually, looks like Arfon started a style guid repo, so we could add the numpy doc string standard somewhere over there: https://github.com/spacetelescope/style-guides

SaOgaz, 2019-01-23:
Posting here for "the record".  I don't think I really have time this quarter to put any work into leading this.

mdlpstsci, 2023-09-05:
Obsolete.  The INS Data Handbooks are the primary source of documentation on the HSTCAL pipeline for users.

closed 2023-09-05 by mdlpstsci