spacetelescope/hstcal #320: CTE related OS dependent runtime floating point precision differences

jamienoss, 2018-06-21:
There exist 2 failing regression tests on Pandokia for WFC3. These only exist on the Linux box. @mdlpstsci checked the input files on both machines (OSX & Linux), they are identicle.

 * ``wf3/onorbit/uvis/calwf3_uv_01.calwf3_uv_01``
 * ``wf3/onorbit/uvis/calwf3_uv_05.calwf3_uv_05``

Diffs found only in CTE corrected output files, ``crc.fits`` & ``flc.fits`` where 56 - 97% pixels differ in the last 1-3 last sig. figs.

The failing tests date back to [3/28/18]() corresponding with #306 and its associated change in the HSTCAL conda build recipe https://github.com/astroconda/astroconda-dev/pull/122.

Looking through the diffs the following might be the culprit:

 https://github.com/spacetelescope/hstcal/commit/fad902e4b3d7aeb80ee741e3b3755e5e48f2ce5f
 where ``rpath``s were added to all ``wscript`` build scripts, ``rpath=bld.env.LIBPATH_CFITSIO``.

In the astroconda recipe the removal of the OS ``[osx]`` specifier for the ``libgcc`` build and runtime dependencies might also be the cause.

More to come...

jamienoss, 2018-06-21:
## Before (ce7fe09)

[03/28/18 pandokia install log](https://ssb.stsci.edu/steuermann/s21.cgi?action=log&name=conda_tests_py3_2018-03-28-10:30:01/nott:conda_testing3/dev_tests) installs ``hstcal=2.0.0.dev0+gce7fe09-2``. When doing a ``conda install hstcal=2.0.0.dev0+gce7fe09`` you get:
```
## Package Plan ##

  environment location: /home/jnoss/.conda/envs/issue320-28th

  added / updated specs: 

    - hstcal=2.0.0.dev0+gce7fe09

The following packages will be downloaded:

    package                    |            build

    ---------------------------|-----------------

    hstcal-2.0.0.dev0+gce7fe09 |                2        26.1 MB  http://ssb.stsci.edu/astroconda-dev

The following NEW packages will be INSTALLED:

    hstcal: 2.0.0.dev0+gce7fe09-2 http://ssb.stsci.edu/astroconda-dev

Proceed ([y]/n)? 
```
There are no installed dependencies as expected from that [build recipe](https://github.com/astroconda/astroconda-dev/blame/b46f46f5c49609017d1e0d97985d0cfb0bbaadf7/hstcal/meta.yaml) (we're bonly concerned about the Linux build):
```yml
...
requirements:
    build:
    - gcc >=4.7 [osx]

    run:
    - libgcc >=4.7 [osx]
...
```
``ldd <path>acscte.e`` yields:
```
linux-vdso.so.1 =>  (0x00007fffa8bfd000)
	libm.so.6 => /lib64/libm.so.6 (0x0000003639a00000)
	libgomp.so.1 => /usr/lib64/libgomp.so.1 (0x0000003269c00000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x0000003639200000)
	libc.so.6 => /lib64/libc.so.6 (0x0000003638e00000)
	librt.so.1 => /lib64/librt.so.1 (0x0000003639e00000)
	/lib64/ld-linux-x86-64.so.2 (0x0000003638a00000)
```

 ## After (d741b49)

[03/29/18 pandokia install log](https://ssb.stsci.edu/steuermann/s21.cgi?action=log&name=conda_tests_py3_2018-03-29-10:30:02/nott:conda_testing3/dev_tests) installs ``hstcal=2.0.0.dev12+gd741b49-3``. When doing a ``conda install hstcal=2.0.0.dev12+gd741b49`` you get:
```
## Package Plan ##

  environment location: /home/jnoss/.conda/envs/issue320-29th

  added / updated specs: 
    - hstcal=2.0.0.dev12+gd741b49

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    curl-7.60.0                |       h84994c4_0         140 KB
    libcurl-7.60.0             |       h1ad7b7a_0         495 KB
    openssl-1.0.2o             |       h20670df_0         3.4 MB
    ca-certificates-2018.03.07 |                0         124 KB
    hstcal-2.0.0.dev12+gd741b49|                4         7.0 MB  http://ssb.stsci.edu/astroconda-dev
    libgcc-7.2.0               |       h69d50b8_2         304 KB
    cfitsio-3.440              |                3         2.8 MB  http://ssb.stsci.edu/astroconda-dev
    ------------------------------------------------------------
                                           Total:        14.3 MB

The following NEW packages will be INSTALLED:

    ca-certificates: 2018.03.07-0                                              
    cfitsio:         3.440-3                http://ssb.stsci.edu/astroconda-dev
    curl:            7.60.0-h84994c4_0                                         
    hstcal:          2.0.0.dev12+gd741b49-4 http://ssb.stsci.edu/astroconda-dev
    libcurl:         7.60.0-h1ad7b7a_0                                         
    libgcc:          7.2.0-h69d50b8_2                                          
    libgcc-ng:       7.2.0-hdf63c60_3                                          
    libssh2:         1.8.0-h9cfc8f7_4                                          
    libstdcxx-ng:    7.2.0-hdf63c60_3                                          
    openssl:         1.0.2o-h20670df_0                                         
    zlib:            1.2.11-ha838bed_2                                         

Proceed ([y]/n)?
```
Now there _are_ installed dependencies as expected from the new [build recipe](https://github.com/astroconda/astroconda-dev/blame/master/hstcal/meta.yaml):
```yml
...
requirements:
    build:
      - cfitsio
      - pkg-config
      - gcc >=4.7

    run:
      - cfitsio
      - libgcc >=4.7
```
``ldd <path>acscte.e`` yields:
```
linux-vdso.so.1 =>  (0x00007ffc553eb000)
	libm.so.6 => /lib64/libm.so.6 (0x0000003639a00000)
	libcfitsio.so.6 => /home/jnoss/.conda/envs/issue320-29th/bin/../lib/libcfitsio.so.6 (0x00007f33cfb3f000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x0000003639200000)
	libcurl.so.4 => /home/jnoss/.conda/envs/issue320-29th/bin/../lib/libcurl.so.4 (0x00007f33cf8ca000)
	libgomp.so.1 => /home/jnoss/.conda/envs/issue320-29th/bin/../lib/libgomp.so.1 (0x00007f33cf6a6000)
	libc.so.6 => /lib64/libc.so.6 (0x0000003638e00000)
	/lib64/ld-linux-x86-64.so.2 (0x0000003638a00000)
	libssl.so.1.0.0 => /home/jnoss/.conda/envs/issue320-29th/bin/../lib/./libssl.so.1.0.0 (0x00007f33cf42f000)
	libcrypto.so.1.0.0 => /home/jnoss/.conda/envs/issue320-29th/bin/../lib/./libcrypto.so.1.0.0 (0x00007f33cefef000)
	libz.so.1 => /home/jnoss/.conda/envs/issue320-29th/bin/../lib/./libz.so.1 (0x00007f33cedd8000)
	librt.so.1 => /lib64/librt.so.1 (0x0000003639e00000)
	libdl.so.2 => /lib64/libdl.so.2 (0x0000003639600000)
```

 ## Summary

The differences are that ``libgcc=7.2.0`` is now being installed as a dependency were it was not before and now new conda installs are linked against for:
 * libcfitsio
 * libcurl
 * libssl
 * libcrypto
 * libz
 * ligomp

The last one, ``libgomp``, could be the cuplrit, only the CTE code usese it, however, I'm not sure why only this was linked to thenew conda install and _not_ the rest of ``libgcc`` which it gets packaged with. I'm also not sure of why a ``libgomp`` change would cause the observed fpp diffs - either I coded something wrong and created a race or there's a bug in that GNU implementation of OpenMP.

jamienoss, 2018-06-21:
Looking at the execs with ``ldd`` and/or `otool -L`` I don't think the issue is with https://github.com/spacetelescope/hstcal/commit/fad902e4b3d7aeb80ee741e3b3755e5e48f2ce5f the addition of the ``rpath``s to the ``wscript``s. I think it's more likely to be the removal of the ``[osx]`` specifier from the recipe.

jamienoss, 2018-06-21:
Adding back the ``[osx]`` specifier to the ``libgcc`` reqs in the build ``meta.yml`` only removes the requirement for ``libgcc``, ``libgcc-ng`` is still a dependency and its ``libgomp`` is linked against. ``libgcc-ng:    7.2.0-hdf63c60_3`` specifically.

jamienoss, 2018-06-21:
> This must be coming from cfitsio or pkg-config (N.B. the latter is only a build dep so shouldn't be the issue).

``pkg-config`` has nothing to do with this.

jamienoss, 2018-06-21:
The [astroconda-dev/cfitsio/meta.yml](https://github.com/astroconda/astroconda-dev/blob/master/cfitsio/meta.yaml) specifiies ``libgcc`` dependencies but again, only for OSX. The inclusion of ``cfitsio`` therefore, shouldn't be the conduit for ``libgcc`` being installed as a dependency. Hmmm...

jamienoss, 2018-06-21:
``cfitsio`` requires ``curl`` which requires ``libgcc-ng`` but not directly as it has no req in [its meta.yml](https://github.com/conda-forge/curl-feedstock/blob/master/recipe/meta.yaml).

There's no need to go any further down that rabbit hole.

Do all ver. of ``cfitsio`` need ``curl``, hmmm?

jamienoss, 2018-06-21:
``cfitsio>=3.430`` (``astroconda-dev``) is the culprit.

jamienoss, 2018-06-21:
Who cares, it's not surprissing that something requires `libgcc``, we do and should also. Well, I guess on Linux it's there, it's just a question of OS shipped vs conda dist.

jamienoss, 2018-06-21:
The PR responsible for the explicit dependcy of ``curl`` with ``cfitsio`` https://github.com/astroconda/astroconda-dev/pull/119

jamienoss, 2018-06-22:
@jhunkeler Hey dude, can you remember the rationale for removing the ``[osx]`` specifier from the ``gcc`` and ``libgcc`` dependencies in the [hstcal meta.yml](https://github.com/astroconda/astroconda-dev/pull/122/files#diff-7e892d3ded9f415ebee4c59e5a4d3bce)?

They're still there for the [cfitsio meta.yml](https://github.com/astroconda/astroconda-dev/blob/master/cfitsio/meta.yaml#L18-L25).

jamienoss, 2018-06-22:
Let's keep this issue to figuring out why the change in ``libgomp`` creates the data diffs (if that really is the culprit) and move the runtime lib dicussion to #321

mdlpstsci, 2018-06-25:
WFC3 has three regression tests involving the CTE correction.  calwf3_uv_01.py and calwf3_uv_05.py both failed, and these tests are processing full frame images.  calwf3_uv_32.py was successful, and it is processing subarray data.

mdlpstsci, 2018-06-28:
I created an hstcal-dev environment on Linux (Nott) with only the necessary dependencies from astroconda-dev.  I also had to install cfitsio and pkg-config.  As previously noted and for completeness, pulling in cfitsio-3.440 also forced some additional packages to be installed: 
 * curl:            7.60.0-h84994c4_0
 * libcurl:         7.60.0-h1ad7b7a_0
 * libgcc-ng:       7.2.0-hdf63c60_3
 * libssh2:         1.8.0-h9cfc8f7_4
 * openssl:         1.0.2o-h20670df_0
 * zlib:            1.2.11-ha838bed_2

I configured the hstcal package to be built without libgomp.  This was verified via "ldd acscte.e" (for example) and the fact the execution of the test took about 10x longer than usual.  In this configuration, the differences between the test output and the truth files persisted.  Boo hoo.

jamienoss, 2018-06-28:
Uh oh.

jamienoss, 2018-06-28:
Maybe it really is a ``cfitsio`` issue.

mdlpstsci, 2018-06-29:
In an attempt to determine if the regression failures can be isolated to cfitsio, I installed the oldest version of cfitsio (3.370) from astroconda in the test environment and ran the regression tests.  This did **NOT** address the incompatibility issue.  We know the last version of cfitsio which generated compatible results on Linux (and OSX) was version 3.280 from 2011!  We then upgraded to version 3.430 circa 2018 to address security issues, and this is when the incompatibilities began on Linux.  I grabbed version 3.280 from the HEASARC website and generated the *.so on Linux and am trying to work with this version.

mdlpstsci, 2018-06-29:
After some discussion with Matt, he built cfitsio-3.280 and published the
package as part of the astroconda channel.  This is the old version of cfitsio
which we KNOW when incorporated as part of hstcal, the WFC3 regression tests
were successful on Linux (as well as OSX). However, I was NOT able to achieve
compatibility with the failing WFC3 datasets under this configuration.  Why?

Even though this old version of cfitsio-3.280 does NOT require curl (which seems
to be the culprit in forcing the use of newer versions of gcc), the latest conda
machinery is in use.  The conda package recipe contains curl/libcurl as an explicit
dependency for cfitsio whether or not it is actually needed, so it seems the gcc/libgcc
version of 7.2 will be used to compile cfitsio-3.280, and our executables will
use libgcc-v7.2.  The curl was introduced as a dependency in cfitsio-3.420.

If we could get rid of this unnecessary dependency on curl and use an older version
of gcc/libgcc (version 4.7?) or the gcc version we were using in Feb 2018, then I
theorize that we should achieve compatibility.  If this were to be the case, then 
we know the newer version of gcc is the culprit, as our system does not use curl. 
If we could use different versions of gcc/libgcc at will, we could be more sure of 
the culprit.  We could also test cfitsio-3.370 as this is another version of cfitio 
in astroconda (yes, I tested with it too).

For completeness, I did some manual building of cfitsio-3.280 on Linux with what I 
believed was an older version of gcc.  I also built hstcal including my version of cfitio
and not the one that is in the conda environment (checked ldd on calwf3.e for example).
None of these many variations achieved compatibility.  I admit that I am not entirely
sure which gcc/libgcc is in play at any instant as I think conda is very clever.  The 
only thing I can think to do now is to work with a conda expert dedicated to this issue 
in the hopes that we can force the build configuration that we want.  Sorry, I am a bit 
fuzzy now after all the variations, and there might be inconsistencies in this logic.

jamienoss, 2018-06-30:
We need to clone the recipe and fix the version of ``cfitsio`` to the older version that was originally shipped with ``hstcal``, also remove the explicit ``curl`` dependency. Build the correct version of ``hstcal`` with ``conda-build`` and install this local version. Then run the tests.

Since we supply the build for ``cfitsio`` its build recipe may also need altering and for us to build it. However, I don't know whether the build and local install of ``hstcal`` will then use this local ``cfistio`` build.

jamienoss, 2018-06-30:
`` conda search cfistio`` 
```
Loading channels: done
# Name                  Version           Build  Channel             
cfitsio                   3.280               0  astroconda          
cfitsio                   3.370               1  astroconda          
cfitsio                   3.370               2  astroconda          
cfitsio                   3.430               0  astroconda          
cfitsio                   3.430               1  astroconda          
cfitsio                   3.440               0  astroconda  
```

jamienoss, 2018-06-30:
A dry run of ``conda install cfitsio=3.280`` on ``nott`` gives no dependencies, so we shouldn't have to touch its recipe.

jamienoss, 2018-06-30:
* clone my repo of ``astroconda-dev`` and ``checkout`` this branch [issue320-cfitsio](https://github.com/astroconda/astroconda-dev/compare/master...jamienoss:issue320-cfitsio).
 * Edit the ``path`` section to point to your local dir of ``hstcal`.
 * checkout d741b49e5eef35f9a37e3c2007b20330e33fd70c for ``hstcal``
 * cd ``<some root>/astroconda-dev/hatcal``
 * ``conda build ./``
 * ``conda install --use-local hstcal``

I think that should work. You'll need to be on ``nott``.

mdlpstsci, 2018-07-18:
I have been pulled from this investigation to address upcoming build changes.  Having said this, one of the last checks I did was to investigate the truth files which I updated on Pembry and Nott on or about 24 April 2018, as well as the regression testing output on or about 29-31 March 2018.  Please note the following discussion is based upon the two regression tests previously mentioned: calwf3_uv_01 and calwf3_uv_03.

Summary
- The original failures on Pembry on 29 March were due to the missing curl library.  The failures on Nott were due to mismatch with the truth files.  
- Once curl was installed on Pembry, all regression tests passed.  When I investigated further, the CRC files **did not actually match**, but comparison of these files was not part of the regression test so this mismatch was not reported.
- On Nott the FLC (and CRC) files did not match.
- I updated the truth files (as generated on my Mac) on Pembry and Nott on or about 24 April.  At this point the FLC and CRC files on Pembry all matched.  Nott continued/continues to fail regression testing.
==> The change associated with cfitsio (static library built into the HSTCAL executables to shared object, forced compilation of cfitsio with curl and an associated newer version of gcc, ???) did cause a change in the test output on both the Mac and Linux, and **not** just on Linux as previously thought.

jamienoss, 2018-10-02:
Breadcrumb: This could be a symptom of #61, see #358 for details.