spacetelescope/hstcal #323: ACS CTE forward model fails when input image has >6 extensions

jryon, 2018-07-20:
The most likely science use case for the ACS CTE forward model is for users to add artificial stars to an FLC/FLT image, and often these contain >6 extensions (15 in the case of full-frame data). The extra extensions contain distortion-related information, and shouldn't affect the forward model. 

When testing this functionality, I get the following error (jd9w05toq_sm2.fits is a renamed FLT):

> WARNING: running CTE forward model onlygit tag: 2.0.0-14-g31c96c4
> git branch: master
> HEAD @: 31c96c4af35d5c12aa097199fb3455217447901d
> Setting max threads to 32 out of 32 available
> Trying to open /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2.fits...
> Read in Primary header from /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2.fits...
> 
> 
> CALBEG*** ACSCTE -- Version 10.0.0 (19-Jan-2018) ***
> Begin    19-Jul-2018 13:13:46 EDT
> Input    /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2.fits
> Output   /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2_ctefmod.fits
> Trying to open /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2.fits...
> Read in Primary header from /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2.fits...
> APERTURE WFC
> FILTER1 F606W
> FILTER2 CLEAR2L
> DETECTOR WFC
> DESCRIP read in as: 
> DESCRIP read in as: 
> DESCRIP read in as: 
> ERROR:    Error processing /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2.fits.
> ERROR:             HSTIO error 114:  Filename /grp/hst/acs3/ryon/ctecorr/forward_mod/jd9w05toq_sm2.fits EXTNAME SCI EXTVER 3 CFITSIO status 301
> ffopen could not move to the specified extension: extension with EXTNAME = SCI, and with EXTVERS = 3, doesn't exist or couldn't be opened.
> Error opening image array.
> *** stack smashing detected ***: /grp/hst/acs3/ryon/ctecorr/hstcal/bindev/acscteforwardmodel.e terminated

@jamienoss determined that when such a file is opened by CALACS, it finds that there are 5 imsets, and fails when it reaches the 3rd because no SCI extension is found (EXTNAME=SCI and EXTVER=3 is never true for ACS data). He also determined that the stack smashing warning comes from the cfitsio error being longer than the buffer allowed for it.

Potentially relevant parts of CALACS:
[https://github.com/spacetelescope/hstcal/blob/2.0.0/pkg/acs/calacs/include/acs.h#L33-L34](https://github.com/spacetelescope/hstcal/blob/2.0.0/pkg/acs/calacs/include/acs.h#L33-L34)
[https://github.com/spacetelescope/hstcal/blob/2.0.0/pkg/acs/calacs/lib/getacskeys.c#L94-L95](https://github.com/spacetelescope/hstcal/blob/2.0.0/pkg/acs/calacs/lib/getacskeys.c#L94-L95)

jamienoss, 2018-07-20:
@jryon thanks for opening this.

jamienoss, 2018-07-20:
There are two issues here:

 * The message buffer isn't large enough for the error returned by cfitsio. This is the cause of the stack protection trigger.
 * Something odd about the number of ``imsets`` being read/computed ([src here](https://github.com/spacetelescope/hstcal/blob/2.0.0/pkg/acs/calacs/lib/getacskeys.c#L94-L95)) and/or the data is funky. Could also be that there isn't always 3 ext per imset/group as is [hard coded here](https://github.com/spacetelescope/hstcal/blob/2.0.0/pkg/acs/calacs/include/acs.h#L33-L34)

jamienoss, 2018-07-20:
This also isn't specific to the forward model, it is at least ``acscte`` but more likely a more widespread issue that, in the past, is normally attributed to user error, i.e. sections of the pipeline being used in ways not intended. However, this is how the forward model piggybacks off of the normal CTE code, thus exposing the pre-existing issue, i.e. it may be being used in a way not originally intended.

E.g. instead of running ``acscteforwardmodel.e jd9w05toq_sm2.fits`` running ``acscte.e jd9w05toq_sm2.fits`` yeilds the exact same error, hence the "at least".

mdlpstsci, 2018-07-23:
The code is now being used in manner not previously anticipated.  An ACS Imset consists of SCI, ERR, and DQ FITS extensions all with the same EXTVER = number as a set.  After AstroDrizzle processing, new WCS-related FITS extensions are appended to the previous data.  Since AstroDrizzle is basically the "end of the line" in terms of processing the data, any issues arising from additional singleton data extensions existing in the FITS products was not seen previously.  The independent code modules of CALACS compute the number of Imsets simply by taking the total number of extensions in a FITS file and dividing by the number of extensions in an ACS Imset.  This simple computation is no longer applicable in today's world.  The true number of actual Imsets needs to be determined in order for the forward model processing to proceed.  The "other" singleton extensions in the FITS file can and should be ignored.

jamienoss, 2018-07-23:
This should be dynamic, i.e. read from the header. Is there a header keyword for this?

jryon, 2018-07-23:
It looks like GCOUNT = 1 in every extension in FLC files from MAST. Would it be possible to use the NEXTEND keyword in the primary header? There should only be two options for FLCs with more than 6 extensions: there should either be 15 (full-frame) or 8 (subarray).

jamienoss, 2018-07-23:
> It looks like GCOUNT = 1 in every extension in FLC files from MAST

I would have guessed so. This issue is pretty massive in its implications!

> Would it be possible to use the NEXTEND keyword in the primary header?

``NEXTEND`` is already used and semantically defined as the number of extensions. We need something for the number of extensions per group/imset. Such metadata cannot exist in the primary header as it is group dependent. This variation is the very cause of this issue in the first place. I.e. at present it is hardcoded in the src code, moving this to the primary header is equivalent.  For it to vary as needed, it must do so per extension header. E.g. principle ext per imset should have this set such that the next ``GCOUNT`` extensions belong to the same group. The ``GCOUNT`` value in those extensions' headers should read 1. 

The above is an example design, I'm not convinced that it is good enough to be acceptable. My concern being the part where the "child" extensions (to the group) have ``GCOUNT=1`` .

mdlpstsci, 2018-07-24:
Just some things to consider...

Are we over-complicating this current problem?  The ACS  Imset by definition has only three extensions (EXTNAME = SCI, ERR, and DQ).  The composition of the Imset, both in type and number of extensions, is instrument-dependent. The EXTVER in each one of these extensions is a way to associate the triple, a triple in the case of ACS, and is the counter used for each Imset present in a FITS file.  We want to know how many Imsets are in the file.  

We want to know (1) the size of the Imset as this is instrument-dependent, and (2) how many Imsets are in any particular FITS file.  
(1) The HSTIO software has in its design the concept of Imset for the various instruments, so the size of the Imset is known by CALXXX according to the functions that have been used to access the data.  For good or bad, this is how HSTIO is written (i.e., getSingleGroup() for ACS versus getSingleNicmosGroup() for Nicmos, 3 and 5 extensions respectively).  

(2) To be general we need to have software which opens the FITS file and determines in real-time how many Imsets are in a file.  This number could then populate a new keyword in the FITS primary header.  One would think this would be enough and subsequent software (e.g., software that would see the keyword is written with a non-zero value for example) then to read this value.  However, there is no guarantee the keyword value is correct (i.e., someone was mucking with the file) so determination might need to be done every time the number of Imsets needs to be known.  As such, I am not sure a new keyword is a solution at all.

Aside: GCOUNT must fulfill this equation
    Nbits = |BITPIX| × GCOUNT × (PCOUNT+NAXIS×...×NAXISm)
and adding a new keyword to the instrument data headers requires some bureaucratic overhead I seem to recall.  Also, GCOUNT carries some baggage (https://archive.stsci.edu/fits/users_guide/node28.html) and may not be appropriate.

These comments are food for thought.  My personal opinion off the "proverbial top of my head" for the short term (or perhaps forever) is to write a discovery routine which determines on-the-fly the number of Imsets in a file.

jamienoss, 2018-07-24:
@mdlpstsci I'm totally with you, this problem is never going to be fixed for the broader scope. We need a workaround.

>For good or bad, this is how HSTIO is written (i.e., getSingleGroup() for ACS versus getSingleNicmosGroup() for Nicmos, 3 and 5 extensions respectively).

10 bucks says there's no generic base func e.g. ``getGroupOfN()`` that ``getSingleGroup()`` and ``getSingleNicmosGroup()`` wrap.

@jryon what is it that you want the forward model to act on, just imsets or...?

jryon, 2018-07-24:
It'd be great if the forward model could run on the SCI extensions of any ACS WFC image. The use case I'm mostly thinking of is FLT/FLCs downloaded from the archive. These have had updatewcs run on them, which appends several extra extensions of distortion information.

Is a workaround possible for 2018.3? It's ok if not, we'd just have to specify in the documentation for the forward model that it can only run on images with one or two imsets (FLT/FLCs or BLV/BLC_TMPs from CALACS).

jryon, 2018-07-26:
I made a wrapper for the forward model for acstools for my own use/edification, and included a workaround for this issue. If a fix for CALACS can't be completed in time for 2018.3, the acstools wrapper could be a (temporary?) solution. At least that way, users could run the forward model on FLCs from MAST from acstools, if not directly with the executable.

I submitted my wrapper as a PR to acstools upon Pey-Lian's suggestion: https://github.com/spacetelescope/acstools/pull/61  No problem if we decide this is not a reasonable way to deal with this issue.

jamienoss, 2018-07-27:
@jryon _c.f._ https://github.com/spacetelescope/acstools/pull/61#discussion_r205780959

Can you check that the issue is really related to ``NEXTEND`` > 6? I can understand that this seems the case by coinidence as there should only be, at most, 2 imsets for ``ACS``, 1 per chip, totalling 3*2 = 6 extensions.

Even though this is unlikely to happen in reality, what would happen if two exposures were mashed into the same file, i.e. there existed 4 imsets? This could trivially just be a copy of the original expected 2.

jamienoss, 2018-07-27:
FYI: I think the solution is to simply count the number of ``SCI`` extensions and then set ``ACSInfo::nimsets`` to this value. I'm just not sure of the cleanest route to finding the total number of ``SCI`` extensions - ideas always welcome.

jamienoss, 2018-07-27:
Is the definition of an imset clearly defined somewhere?

jryon, 2018-07-27:
The issue is not only for NEXTEND > 6. I ran the test @jamienoss suggested, with two sets of the same two imsets in one FITS file. This file doesn't contain any distortion extensions. I set NEXTEND=12 and the EXTVER keywords in the copied pair of imsets to 3 and 4, respectively. The code did not crash and the output looks right. 

I also tested appending a copy of [SCI, 1] alone to the end of the file, updating NEXTEND=7 and EXTVER=3. The code didn't crash, but it also didn't run on the copy of [SCI, 1], and the output file didn't have a 7th extension.

If I set NEXTEND incorrectly in either of the above files (e.g., NEXTEND=15), the code crashes with the same error as before. The error specifies the first SCI, EXTVER it expected to find but couldn't (which was 5 for the 4 imset image).

jamienoss, 2018-07-27:
@jryon awesome, thanks for doing this. 😃

jryon, 2018-07-27:
Imsets are discussed here: http://www.stsci.edu/hst/HST_overview/documents/datahandbook/intro_ch23.html#1913215

jamienoss, 2018-07-27:
> Imsets are discussed here: http://www.stsci.edu/hst/HST_overview/documents/datahandbook/intro_ch23.html#1913215

This is interesting as it doesn't define it to be specific to a ``SCI``, ``ERR``, & ``DQ`` group but simply as a group of extensions associated by ``EXTVER``. With this definition, the total number of imsets is just ``max(EXTVER)``. This definition seems pretty degenerate and ambiguous. I would also be inclined to consider it a semantic redefinition/overloading of ``EXTVER`` and therefore possibly "fits illegal". 

Here's the [official FITS standard (4.0)](https://fits.gsfc.nasa.gov/standard40/fits_standard40aa.pdf) definition:
```
EXTVER keyword. The value field shall contain an integer to be
used to distinguish among different extensions in a FITS file with
the same type and name, i.e., the same values for XTENSION and
EXTNAME. The values need not start with 1 for the first extension
with a particular value of EXTNAME and need not be in sequence
for subsequent values. If the EXTVER keyword is absent, the file
should be treated as if the value were 1.
```

The official def of ``EXTVER`` is used to disambiguate between extensions with identical ``EXTNAME`` and contains no meta-semantics of an association.
E.g.

 ``EXTNAME``  | ``EXTVER``
:---: | ---:
SCI |  1
ERR |  1
DQ  | 1
SCI |  2
ERR |  2
DQ  | 2
BLAH | 1 
BLAH | 2

In the above, there are two image sets and two other data images named "BLAH". Because there are two of them they need to use ``EXTVER`` to disambiguate themselves, however, they are then implicitly associated with each of the two imsets as they share the same ``EXTVER``. 

Perhaps I have this wrong and the format is for the "BLAH" exts to have ``EXTVER`` = 3 & 4?

mdlpstsci, 2018-07-27:
The concept of Imset is purely an STScI convention used for the "new" instruments at the time.  The ACS Imset is SCI, ERR, and DQ.  The NICMOS Imset is SCI, ERR, DQ, SMPL, and INTG.  It is simply a way that was used to organize or conceptually associate the three (or five) data extensions as a group. As you note, EXTVER "is used to disambiguate between extensions with identical EXTNAME".  The Imset convention has overloaded the use of EXTVER, but this does not violate the FITS standard.

The Imset concept is not perfect, but it is based upon the defined data extensions for science instruments.   In your above example, the BLAH extensions should not be associated with the Imsets just according to the EXTVER values of "1" and "2" because they are not part of the full data description for the instruments.  Understandably, a user *unfamiliar* with the science data could be misled.  Using the EXTVER values of "3" and "4" for BLAH adds some clarity to this foggy notion.

jamienoss, 2018-07-27:
>The concept of Imset is purely an STScI convention

Sure, but it doesn't mean that it doesn't have to be a complete definition, right?

> Using the EXTVER values of "3" and "4" for BLAH adds some clarity to this foggy notion.

But _must_ they be? Is it required by a definition of a design standard etc? If not, then they can still be 1 & 2 and thus the def of an imset of being an overloading of ``EXTVER`` is an incomplete definition.

What's also interesting is that the FITS design def implies that the ``EXTNAME`` is the object/root and that ``EXTVER`` be meta to this, whilst the [ST def](http://www.stsci.edu/hst/HST_overview/documents/datahandbook/intro_ch23.html#wwfootnote_inline_138) seems like it's the complete opposite:
```
The keyword EXTNAME is used to specify the extension names of different images in the same imset.
```
I.e. the ``EXTVER`` is the root object and ``EXTNAME`` the meta augmenting it.

jamienoss, 2018-07-27:
> This is interesting as it doesn't define it to be specific to a SCI, ERR, & DQ group but simply as a group of extensions associated by EXTVER. With this definition, the total number of imsets is just max(EXTVER).

This is only true, however, if the set of ``EXTVERS`` is continuous, i.e. 1, 2, 3.. and not 1, 3, 5, 9 etc. I haven't found any design spec requiring this. Given the official FITS std def of ``EXTVER`` there doesn't need to be one.

Without one, the definition becomes:
```python
nimsets = len(set([hdus.ver for hdus in hduList]))
```
The above actually covers both definitions so is still complete.

jamienoss, 2018-07-30:
Another thought that's just come to me whilst trying to figure this out - an extension with an empty ``EXTVER`` keyword, how are these written out? If the writing func gets passed an ext without an explicit ``EXTVER`` it's ok for it to write to file as such, but only if it's the 1st occurrence in the file, otherwise it has to have unique ``EXTVER``.  The write func would have to check the other extensions to determine this though - how does that get designed/implemented? If the file is being written out from scratch, then it could simply inc. a counter on every ext write, even ensuring that no ``EXTEVER`` is empty. However, if appending an extension to a file uniqueness must be determined to avoid clashes.

Since an imset is only an ST convention, the latter point doesn't actually have to be true as this clash is completely acceptable by the FITS std.

jamienoss, 2018-07-30:
I guess the hole in the def is that an ext with an empty ``EXTVER`` is simply not an imset. I haven't seen this explicitly defined anywhere yet though.

jamienoss, 2018-07-30:
Even ``CFITSIO`` seems confused about implementing the FITS std of ``EXTVER``:

[CFITSIO v3.4: HDU Access Routines](https://heasarc.gsfc.nasa.gov/docs/software/fitsio/c/c_user/node36.html)
```c
int fits_movnam_hdu / ffmnhd
      (fitsfile *fptr, int hdutype, char *extname, int extver, > int *status)
```
```
If the input value of extver is 0 then the EXTVER keyword is ignored and the first HDU
with a matching EXTNAME (or HDUNAME) keyword will be found.
```
The FITS std allows for ``EXTVER == 0`` and therefore the above ``CFISTIO`` func has a hole in its functionality space since an extension with ``EXTVER = 0`` doesn't have to be the 1st extension and therefore the above func cannot be used to find it.

jamienoss, 2018-08-03:
Possible fix over at #330

PR #330 (2018-08-06): Use findTotalNumberOfImsets() to compute ACSInfo::nimsets
  Fixes #323
commit bebc9ca4 (2018-08-06): Use findTotalNumberOfImsets() to compute ACSInfo::nimsets

Fixes #323
PR #335 (2018-08-09): Make acs->nimsets = findTotalNumberOfImsets() conditional on forwarModelOnly
Fixes regression introduced from #330

Includes refactoring of ``GetACSKeys()`` into 2 sep funcs
``getACSKeys()`` & ``checkACSKeys()``, the former doing only the literal
getting.

Fixes #333
Resolves #323
commit b22724ba (2018-08-09): Make acs->nimsets = findTotalNumberOfImsets() conditional on forwardModelOnly

Fixes regression introduced from #330

Includes refactoring of ``GetACSKeys()`` into 2 sep funcs
``getACSKeys()`` & ``checkACSKeys()``, the former doing only the literal
getting.

Fixes #333
Resolves #323
closed 2018-08-06 by mdlpstsci