spacetelescope/hstcal #324: Convert string keyword values to upper-case at point of ingest for calacs

mdlpstsci, 2018-07-26:
String keyword values should be converted to upper-case in the routine(s) which is(are) the point of ingest.  In this way the conversion need only be done once and the other code modules do not need to do any checking or further conversion.  Care will need to be taken to not break existing code.

jamienoss, 2018-07-27:
I thought the canonical convention was to make everything lower case?

mdlpstsci, 2018-07-27:
A convention should be chosen, lower- or upper-case, and then used consistently.  I tend to convert to upper-case for my comparisons, but your question raised my curiosity.  In an albeit brief search, I found the upper-case convention to be preferred (or so the claim asserted based upon statistics), and I found one instance in dealing with a foreign language where only the upper-case conversion yields the "correct" answer (German Eszett).  (Note: this discussion ignores case-insensitive comparisons.) It is good to be sensitive to this issue, but CALACS is a limited world where I have noticed case inconsistency in the string values of the HST FITS keywords which are all in English.

pllim, 2018-07-27:
I am not sure what prompted this but FITS is case-insensitive, so it doesn't matter if things are lower or upper case.

jamienoss, 2018-07-27:
>I am not sure what prompted this but FITS is case-insensitive, so it doesn't matter if things are lower or upper case.

Actually, is this true as I just found this when looking for something else ([src](https://heasarc.gsfc.nasa.gov/docs/software/fitsio/c/c_user/node52.html), section 6:
```
Test that the keyword name contains only legal characters (A-Z,0-9, hyphen, and underscore) or that the keyword record contains only legal printable ASCII characters
  int fits_test_keyword / fftkey (char *keyname, > int *status)

  int fits_test_record / fftrec (char *card, > int *status)
```
If it were case insensitive then the ``[A-Z]`` group would be ``[a-zA-Z]``.

mdlpstsci, 2022-01-12:
This was just a convention and a preference, but it is not an actual bug at this time.

closed 2022-01-12 by mdlpstsci