spacetelescope/hstcal #329: Extend the bias shift corretion algorithm of CALACS to handle the WFC 512 and 1K subarrays

mdlpstsci, 2018-08-01:
This is a place-holder issue to extend the bias shift correction in the BLEVCORR step of the ACSCCD algorithm of CALACS to handle the WFC 512 and 1K subarrays.  Currently, the algorithm depends upon the data having both physical and virtual overscan.  The virtual overscan is not present in the 512 and 1K subarray data so the team needs to define an alternative value to be used for the algorithm.

jryon, 2025-04-15:
Can someone confirm if the bias shift correction is not currently enabled for 512 and 1K subarray data? The ACS DHB says it is not currently enabled, but the fact that #312 is completed makes me second guess it all.

mdlpstsci, 2025-04-15:
The only data supported is the following:

"WFC1A-2K", "WFC1B-2K", "WFC2C-2K", "WFC2D-2K",
"WFC1-IRAMPQ", "WFC1-MRAMPQ", "WFC2-MRAMPQ", "WFC2-ORAMPQ",
"WFC1-POL0UV", "WFC1-POL0V", "WFC2-POL0UV", "WFC2-POL0V",
"WFC1-POL60UV", "WFC1-POL60V", "WFC2-POL60UV", "WFC2-POL60V",
"WFC1-POL120UV", "WFC1-POL120V", "WFC2-POL120UV", "WFC2-POL120V"

pllim, 2025-04-15:
@mdlpstsci , is that answer from the "keyword rules" file you sent to me and Robert this morning? Thanks.

mdlpstsci, 2025-04-15:
@jryon @pllim 
The response is from the doccd.c source code.  Near the top of the routine is an array of subApertures.  There is a large comment above the variable with some description.  Also, at Line 390 there is more description.

This is the routine that I want to re-arrange which we can discuss at another time.

pllim, 2025-04-15:
Ah, I see. Thanks!

https://github.com/spacetelescope/hstcal/blob/1cecb633c5f39b3faf7b653167ae4c82e44ec092/pkg/acs/lib/acsccd/doccd.c#L66-L70

https://github.com/spacetelescope/hstcal/blob/1cecb633c5f39b3faf7b653167ae4c82e44ec092/pkg/acs/lib/acsccd/doccd.c#L387-L427

jryon, 2025-07-02:
@pllim Following up on this from the team meeting. Please use the mean of a portion of the physical prescan instead of the "magic square" for the fat zero estimate. The five columns and the first row of the prescan closest to the amplifier should be thrown out before taking the mean, as there is a bias settling feature in those columns/rows that would skew the mean. In python with numpy, this would be:
```
amp A:  np.mean(sci[:-1,5:24])
amp B:  np.mean(sci[:-1,-24:-5])
amp C:  np.mean(sci[1:,5:24])
amp D:  np.mean(sci[1:,-24:-5])
```
I believe the above should be the same for 1k and 512 subarrays.

nagrogin, 2025-07-02:
The idea would be to fork the 'pedestal' measurement:
*if* the image has a magic square, retain the status-quo of pedestal=mean-of-the-'magic square'; 
*else* pedestal=mean-of-the-truncated-prescan as noted above

pllim, 2025-07-03:
@mdlpstsci , the magic square stuff is pretty embedded in existing code, plus a hardcoded magic numbers:

https://github.com/spacetelescope/hstcal/blob/85c4cf2b6ac7cf94d03bd53c0b27370b31996b27/pkg/acs/lib/acsccd/blev_funcs_postsm4.c#L59

I think it might be cleaner if we create a new file (`blev_funcs_postsm4_subarr_512_1k.c` ❓ ) and do 512 and 1k processing separately. What do you think?

jryon, 2025-07-07:
I ran some tests of full-frames, and found that the magic square bias level is at most ~1-2e- greater than the mean of the first 512 or 1k prescan rows (excluding the first row and first 5 columns). This is small enough to be negligible to the bias shift correction, so please move forward with using the prescan for the bias shift for 512 and 1k subarrays, as described in the previous comment.

pllim, 2025-07-07:
@jryon et al.,  just to be clear, ACS Team has no intention to give user a special option to opt-out from *this new calculation*, right?

nagrogin, 2025-07-07:
@pllim , you are correct.
This will be treated more like BIASCORR than PCTECORR, for the subarrays.
E.g., it will always be done (on all post-SM4 subarray data containing prescan) rather than allowing an opt-out and/or a standalone code outside CALACS.

nagrogin, 2025-08-06:
> @nagrogin<https://urldefense.com/v3/__https://github.com/nagrogin__;!!CrWY41Z8OgsX0i-WU-0LuAcUu2o!zEHBKV2PgIfkCxNb8hgbVpRlaUkjPUNGSTC5nNoS-zCCUvXjPkNuPJbaGF4nX3hkJ0Isgw2hXI5C1Qb-FV7-19iQiQ$> or @jryon<https://urldefense.com/v3/__https://github.com/jryon__;!!CrWY41Z8OgsX0i-WU-0LuAcUu2o!zEHBKV2PgIfkCxNb8hgbVpRlaUkjPUNGSTC5nNoS-zCCUvXjPkNuPJbaGF4nX3hkJ0Isgw2hXI5C1Qb-FV7VcnrDyw$> , do we care about "available but unsupported modes" below?
>
>   *   WFC1-512
>   *   WFC1-1K
>
> Ref: https://hst-docs.stsci.edu/acsihb/chapter-7-observing-techniques/7-7-acs-apertures#id-7.7ACSApertures-table7.7

The legacy subarrays should not be included, as they have different readout timings.

Thanks,

--Norman

pllim, 2025-08-06:
@jryon , regarding your analysis in https://github.com/spacetelescope/hstcal/issues/329#issuecomment-3045870723 , how different was that value from MEANBLEV that is already calculated here?

https://github.com/spacetelescope/hstcal/blob/6ef65399e99f7f41f815bda98a93c9580cc835df/pkg/acs/lib/acsccd/doblev.c#L305

I am trying to see if I can reuse some existing result instead of duplicating work. If that is not the same, I can always write new calculations but just want to make sure. The existing MEANBLEV already seems to go through some fitting and outlier rejection.

pllim, 2025-08-08:
@nagrogin or @jryon , are you able to review https://github.com/spacetelescope/hstcal/pull/685 ? Does it look about right? Let me know if you need help installing from the branch, or want me to help you process anything with it. Thanks!