spacetelescope/hstcal #331: CDELT1/CDELT2 updates for WFC3 and ACS

stscicrawford, 2018-08-07:
SCSB is to supply the appropriate WCS keywords that should be added to NPOLFILE that are accurate and sufficient that the files are compliant with the FITS standards such that they pass fitsverify without any warning. 

cc @stsci-hack @nden @mdlpstsci

mdlpstsci, 2018-08-08:
The implicit statement here is that SCSB will work with the instrument teams to determine the appropriate WCS _keyword_ values in a "do no harm" manner to mitigate any inappropriate interpretation of the data in the reference files.  Since AstroDrizzle uses the CDELT1/CDELT2, it would be useful to understand what default values the software uses for the remaining WCS keywords - under the assumption the software actually needs the values from the remaining WCS keywords for its computations.

pllim, 2026-08-25:
All the people in this issue either retired or left, so closing without further action.

closed 2026-08-25 by pllim