spacetelescope/hstcal #340: TST: Move the rest of active pandokia tests to Jenkins + Artifactory

pllim, 2018-08-21:
Move the following tests over:

- [ ] ACS
- [ ] WFC3

Note: STIS tests are already handled by spacetelescope/stistools#48 by @SaOgaz .

pllim, 2018-09-13:
@mdlpstsci , did your #349 completely address this? I see that you have already opened up separate follow-up issues. In another word, can we stop running pandokia tests yet for HSTCAL?

mdlpstsci, 2018-09-13:
Yes, we can stop running Pandokia tests for ACS and WFC3 with the understanding:
(1) Please do not remove any of the input or truth files on the test machines, and
(2) Some of the tests are not yet running with Jenkins/Artifactory as they are marked as xfail.

pllim, 2018-09-13:
@stscicrawford , do you have a preference whether pandokia should be out of the picture now or later?

stscicrawford, 2018-09-13:
Tests have been migrated, and issues have been opened for the few that need to be further investigated..  I'm happy to have this closed and pandokia turned off.

pllim, 2018-09-13:
@rendinam has turned them off but we won't know till tomorrow's run. :smile: :crossed_fingers:

pllim, 2018-09-14:
Confirmed that HSTCAL no longer runs on pandokia. 🎉

closed 2018-09-14 by pllim