spacetelescope/hstcal #347: Do not add extra error to ERR extensions in ACS CTE forward model

jryon, 2018-08-28:
According to @jamienoss in [#330](https://github.com/spacetelescope/hstcal/pull/330#issuecomment-410352332),

> The CTE correction simply adds 10% of the correction for the error induced, what this should be for the forward model I have no idea.

Ideally, the ERR extensions should be the same before and after forward modeling, so no extra error should be added by the CTE forward model. This is because the input data into the forward model may vary in terms of how many calibration steps have been performed, and so it may not be accurate to add more noise. We'd like to simply simulate the effects of readout on the SCI extensions, and leave any ERR extension updates to the user.

jamienoss, 2018-08-28:
>Ideally, the ERR extensions should be the same before and after forward modelling

I don't think this is correct, right? Any operation introduces an error, even if just real noise. In this case, it's definable. Besides, any model/simulation that didn't yield an error would be un-scientific, surely?

> and leave any ERR extension updates to the user.

This "update" would still have to be functionally conducted by the forward model. Whilst it's now only 10% and thus trivial for the user to add themselves after the fact, it doesn't always have to be. Separating the two isn't great design. It's just as trivial for the user to subtract the current 10% as it is to add it. In addition, the original input is not altered and therefore the user can always use that ERR extension.

jamienoss, 2018-08-28:
> This is because the input data into the forward model may vary in terms of how many calibration steps have been performed, and so it may not be accurate to add more noise.

Sorry, I don't follow this.

hcferguson, 2018-08-29:
Propagation of errors can be subtle. A couple of general principles can avoid rampant confusion:

1.  Don't include systematic uncertainties in error arrays. Document them.
2.  Be sure to discuss in the documentation any operation that might introduce strongly correlated uncertainties in the output if you aren't going to have a covariance output (we generally don't).

The rationale for the first principle is that it is very common for users to combine data sets (e.g. for multiple dithers). Standard propagation of uncertainties generally assumes uncorrelated errors. Systematic errors are correlated.  The JWST pipeline will be carrying around multiple separate uncertainties for this reason, but we never did that for HST.

(Following these principles, I'm not sure I would favor adding in the 10% for the backward CTE correction either.)

jamienoss, 2018-08-29:
For ref. here's the [src](https://github.com/spacetelescope/hstcal/blob/master/pkg/acs/calacs/acscte/dopcte-gen2.c#L270-L278)
```c
...
delta = (PixColumnMajor(cteCorrectedImage->sci.data,k,m) - 
PixColumnMajor(smoothedImage.sci.data,k,m));
threadCounts += delta;
threadRawCounts += Pix(ampImage.sci.data, m, k);
temp_err = 0.1 * fabs(delta);
Pix(ampImage.sci.data, m, k) += delta;

float err2 = Pix(ampImage.err.data, m, k);
err2 *= err2;
Pix(ampImage.err.data, m, k) = sqrt(err2 + temp_err*temp_err);
...
```

hcferguson, 2018-08-29:
If this isn't changed, I'd recommend adding a caution to the documentation. If the systematic uncertainty in the CTE correction is really systematic, it won't beat down as sqrt(N) as you add dithered frames. So users should be backing this out before combining dithered frames and then adding it back in quadrature with the random uncertainties after they do the stacking.

jamienoss, 2018-08-29:
@hcferguson Confused, I now am.

Aren't you stating that for any given image the ERR array is not accurate, i.e. that it isn't the correct or complete representation for its associated SCI array? I would have thought this rendered the point of the ERR array useless, especially if sometimes it is updated/corrected and sometimes isn't. So my first question is, what is it for?

This error is correlated to the correction, it is currently just 10%. If this is not to exist in the ERR array but instead only documented doesn't it become non-trivial to compute after the fact? For the sake of a more clearer argument ignore the simulation but instead consider the CTE correction itself, which is a stage of the pipeline. To comp. this after the fact you'd need to have the input image to the CTE code, which is the output from the stage before the CTE correction. I would have thought this to be a completely unnecessary data dependency that the ERR array was designed to solve by definition, no?

Taking this last point further, if each and every stage is to only have its error contribution documented and not computed, on the fly as a part of the pipeline, are you saying that the end user is supposed to figure out which stages were or were not performed and in which order, and have each of the intermediate images (which isn't the default for all stages) to then compute the resultant error?

jamienoss, 2018-08-29:
@nagrogin any thoughts?

jamienoss, 2018-08-29:
BTW this is not really my domain, so I'm cool with changing it to whatever, as long as @nagrogin and/or @mdlpstsci are cool with it. Just wanted to point this out.

I would like to understand it though.

hcferguson, 2018-08-29:
Yes, by and large the ERR arrays in HST data are not accurate and careful astronomers have to propagate uncertainties on their own. For imaging it does the least harm, though, to carry just the uncertainties of the background at the sky level computed from the CCD noise model (i.e. taking into account read noise, but not Poisson noise due to the sources themselves). This is what astrodrizzle does, I'm pretty sure.

I agree that adding the 10% is cumbersome/impractical after the fact. I just don't think adding 10% to the ERR array helps. 

In the case of the backward correction, I think probably people could compare their photometry on data before and after the correction, take the ratio as the size of the correction and 10% of that as the part of the uncertainty due to CTE.

I suppose one could do the same for the forward modeling. But there are two possible ways to think of the purpose of forward modeling: (1)  "if the CTE model is perfect, what does it do to my photometry?" or (2) "If the CTE model is imperfect to the level thought (10%) how does this propagate into my photometric uncertainties?" For (1) you don't need to propagate errors. 

For (2), you may be okay as long as you are doing just one dither, but otherwise are probably going to get it wrong.  If I were doing this type of analysis I might be inclined to take the brute-force approach and draw the CTE correction from a probability distribution -- always using the same value the same for all my dithers on each realization...and then run this a hundred times, redoing the photometry with whatever technique I used on the real data. Cumbersome and there are probably shortcuts.

jamienoss, 2018-08-29:
> I agree that adding the 10% is cumbersome/impractical after the fact. I just don't think adding 10% to the ERR array helps.

I'm not sue if I completely follow your last point here because I feel like the discussion is being too heavily tied to specific values, i.e. too explicit and not abstracted enough. If the CTE error contribution was non-trivial, wouldn't it _have_ to be added to the ERR array on the fly? And if so, what is the design rationale to not doing so regardless? The fact that it is _currently_ only 10% should be irrelevant, right? I.e. the design _should be_ value agnostic.

> (1) "if the CTE model is perfect, ... For (1) you don't need to propagate errors.

Isn't this only true because a "perfect" model has zero error by definition?

>  If I were doing this type of analysis I might be inclined to take the brute-force approach and draw the CTE correction from a probability distribution

Do you perhaps mean "CTE correction _error_"? If not, isn't this well beyond this issue and instead discussing the entire CTE correction technique itself?

jamienoss, 2018-08-29:
@hcferguson thanks for the further explanation, I haven't dealt with errors since undergrad, now over a  decade ago. I follow what you're saying but am unfortunately still hooked up on a single point - your uncertainties, e.g. 2%, the error is a function of only the image pixel values themselves, nothing else. I understand why you might not wish to add this to the error array. However, the error on the CTE correction is not 10% of the pixel value, it's 10% of the correction, i.e. ``0.1 * |before - after|``.

But anyhow...

jryon, 2018-08-29:
After talking to @nagrogin yesterday and again today, we're still in agreement that we'd like the forward model to refrain from adding error to the ERR arrays.

The reasoning is pretty much option (1) above from @hcferguson. We expect the forward model to be used to study the effects of CTE on photometry using artificial star tests. The data that goes in to the forward model should be equivalent to a BLC_TMP, which, in principle, is the completed exposure not yet read out. In this case, the forward model is assumed to be a perfect simulator of CTE effects during readout. The output from the forward model is like a BLV_TMP, and therefore the content of that image's ERR array should conform to that of a "real" BLV_TMP (one that has simply been processed through acsccd). 

Users may generate their own synthetic data to run through the forward model. Or they might start with an FLC downloaded from MAST, do some processing to back out the flat and dark correction, and then run the forward model on the BLC_TMP-like frame. I'm writing a notebook to guide users with both methods and then running the forward model. I will be including a step immediately after the forward model that zeroes out the ERR arrays and repopulates them as if they were in a BLV_TMP. So, if users care enough to have their ERR arrays be strictly correct, it doesn't really matter what the forward model adds to the ERR array. But, it would likely be best to default to not add any error to the ERR array, since the forward model is intended to simulate readout exactly.

nagrogin, 2018-08-29:
Hi all,

Henry Ferguson wrote:
>
> Sounds sensible.
>
> Question: Since there are systematic uncertainties in the model, is 
> there a way to have reference files for "high" and "low" corrections 
> that people could use to feed the forward model to see how much the 
> model uncertainties might matter? All they have to go on right now is 
> to compare the results with and without the correction.
>

Spoofing the observation date in the header should
modulate the amplitude of the correction --- according
to the modeled time-dependence of the CTE degradation.

Adjusting the background level of the image will
also modulate the amount of CTE correction, based
on the charge-trap-density profile vs. intensity.

Not sure these sorts of things are what Harry has
in mind.  I'm pretty sure that Jay never provided
upper/lower envelopes to his fittings of the trap-density
profile, charge release profile, and long-term time-dependence.
These envelopes would presumably be somewhat correlated.

Cheers,

--Norman

nagrogin, 2018-08-29:
James Noss wrote:
>
> However, I have to confess that I'm completely dumbfounded by the 
> agreement for that embodied by the following rationale:
>
>     ...it would likely be best to default to not add any error to the
>     ERR array since the forward model is intended to simulate readout
>     exactly.
>
> The reality of it is that the model /does not/ simulate the readout 
> exactly. But hey!
>

I fear a semantic disconnect is happening, here.

The first thing anyone is going to do with the output of
the simulator is to feed the faked-up BLV_TMPs through
CTECORR and onwards through CALACS.

My rationale is that the faked-up BLV_TMP ERR array
should look indistinguishable to the ERR array of an
actual BLV_TMP.  Which is easiest to create ab initio,
rather than trying to have the simulator-module worry
about this.

Once that faked-up BLV_TMP gets corrected by CTECORR
on its way through through CALACS, it will have the 10%
"model-uncertainty-noise" added in just like the authentic
BLV_TMPs.

Cheers,

--n

mdlpstsci, 2018-08-30:
It is understood the ACS team would like to have this functionality available as soon as possible.  At this time, the HSTDP-2018.3 (HSTCAL-2.1.0rc2) release is still in testing.  If there are problems found and a new release candidate must be prepared, then we can integrate this update into the source.  If there is no need for another release candidate, then this will be slated for another operations release.  This would be a possible quick fix (HSTDP-2018.3a) or an Astroconda package that would contain the entire delivery to DPAS/DSB with an upgraded HSTCAL that would contain this fix so the public can access the improved functionality.  Some of this commentary is contingent on the testing schedule of DPAS/DSB and what they can accommodate.

jamienoss, 2018-08-30:
@nagrogin 
I totally get what you guys want to do with it. I also understand that perhaps it is not desirable to see signitures of the CTE trails in the ERR array for particular use cases. However, what I don't understand is why this isn't just "hard luck" and the nature of the game?

> My rationale is that the faked-up BLV_TMP ERR array
should look indistinguishable to the ERR array of an
actual BLV_TMP. 

Sure, for what _you_ want to do with it. The issue here is that it's not a "faked-up BLV_TMP" generator but a CTE simulator in public pipeline code. Not including this error on the grounds of never including systematics is one thing, the above however is a whole different ball game.

But, anyhow, I think I've caused us to debate this to death... Peace out ✌️

nagrogin, 2018-08-30:
On 8/30/18 9:56 AM, James Noss wrote:
> @nagrogin <https://github.com/nagrogin>
> I totally get what you guys want to do with it. I also understand that
> perhaps it is not desirable to see signitures of the CTE trails in the
> ERR array for [...]

I am putting the brakes on all this.
Do not make any changes to the existing CALACS code
without my explicit authorization.

Why are we needing to change the ERR-array handling
of the CALACS code, at all?

I shouldn't care what the CTE simulator in the public pipeline
code does to the ERR array of the output for the simulator
(rather than the corrector) case --- our simulator wrapper
will be wiping that ERR array, and resetting it to what we
desire.

Can someone explain to me the reason why we need to be
fiddling with the ERR-array handling, within the CALACS code?

Cheers,

--n

nagrogin, 2018-08-30:
On 8/30/18 8:37 AM, mdlpstsci wrote:
> It is understood the ACS team would like to have this functionality 
> available as soon as possible. 

As per my last email, belay this until I give word.

--Norman

nagrogin, 2018-08-30:
On 8/30/18 1:22 PM, James Noss wrote:
> If by /"our wrapper"/ you mean that the ACS team has their own code 
> that wraps the CTE forward model computation that then "wipes" the ERR 
> array, resetting it to what you
> desire - then I'm totally confused as to why this issue is open?

Exactly my point.

Jenna is creating a jupyter notebook for the simulator procedure,
and the notebook will be doing the ERR-array whitewashing after
running the ctecorr simulator part.

I have spoken with Jenna about closing out the issue.

In the fullness of time, hopefully this year, we will be making a
thorough study of the data-impacts of the CTE correction.  And
hopefully the data will guide us about what is the proper thing
to be doing (other than that ad-hoc 10%-of-the-delta kludge).

But that is still some ways off, and beforehand I don't want to be
fiddling with CTECORR standard operating procedure of several years.

--Norman

jryon, 2018-08-30:
I'm requesting that this issue be closed without updating CALACS at all, so please do not merge PR #348. Any ERR array changes can be handled by the user with guidance from my forward model notebook. We will make it explicitly clear in the notebook that `0.1*|before - after|` is added to the ERR array by the forward model so that if the user does not follow our guidance to recalculate the errors themselves, they will know what is going in to the ERR array.

closed 2018-08-30 by jamienoss