spacetelescope/hstcal #356: Reference file location

stscicrawford, 2018-09-19:
In order to run the software, the environmental variables for the reference files need to be set.  This should be added to the README for running internally and externally.

jaytmiller, 2018-09-19:
The controlling environment variables are CRDS_PATH which defines the cache location and CRDS_SERVER_URL which defines which of 8 strings of servers to use.

The defaults are configure make JWST work in a passive serverless zero configuration way:

CRDS_PATH=/grp/crds/cache
CRDS_SERVER_URL=https://crds-serverless-mode
CRDS_OBSERVATORY=jwst

The default observatory only affects the behavior of secondary CRDS features.    Offsite,  CRDS for JWST should be configured like this:

CRDS_PATH=$HOME/crds_cache
CRDS_SERVER_URL=https://jwst-crds.stsci.edu

In the offsite configuration CRDS_OBSERVATORY is effectively inferred from the server URL.   There is a directly analogous configuration for HST and the default cache works for both projects.   However,  CRDS cannot always guess observatory so setting CRDS_OBSERVATORY for hst can become necessary.

stscicrawford, 2018-09-19:
In addition to those, the following keywords need to also be set for the pipeline software to access the cache:

iref =/grp/hst/cdbs/iref/
jref= /grp/hst/cdbs/jref/

pllim, 2018-09-19:
Hmm I thought this is all covered in Data Handbooks.

mdlpstsci, 2022-01-11:
**_All items JWST are not part of this repository._**

There are actually more HST reference file items which need to be specified for HSTCAL:
crrefer=/grp/hst/cdbs/
iref=/grp/hst/cdbs/iref/
jref=/grp/hst/cdbs/jref/
jtab=/grp/hst/cdbs/jtab/
lref=/grp/hst/cdbs/lref/
mtab=/grp/hst/cdbs/mtab/
nref=/grp/hst/cdbs/nref/
ntab=/grp/hst/cdbs/ntab/
oref=/grp/hst/cdbs/oref/
otab=/grp/hst/cdbs/otab/
uref=/grp/hst/cdbs/uref/

As noted, this is fully documented in the handbooks.

pllim, 2022-01-11:
"cdbs" is retired by RedCats. It is "trds" now.

closed 2025-06-10 by pllim