spacetelescope/hstcal #357: WFC3 inconsistency in CTE processed data when using single or multiple threads on Linux

mdlpstsci, 2018-09-21:
Using the Astroconda HSTDP-2018.3 distribution on a Linux system to process WFC3 data produced inconsistent output (FLC files) when the datasets were processed using a single or multiple threads.  There are instances when the FLC files are the same (via fitsdiff), but the discriminant between the consistent and inconsistent datasets has not been determined at this time.

mdlpstsci, 2018-09-24:
Using the Astroconda HSTDP-2018.1 distribution (HSTCAL 2.0.0) and running calwf3.e on dataset iacr51ohq in the default mode (multiple threads) and then running calwf3.e and forcing the use of a single thread produced no differences in the output FLC files (fitsdiff with no tolerances).  This particular dataset was found to have inconsistent output FLC files when processed with HSTDP-2018.3 single vs multiple threads.  While additional datasets should be processed for better statistics, this result implies the Linux single/multiple thread issue is new for HSTDP-2018.3.

jamienoss, 2018-10-02:
See https://github.com/spacetelescope/hstcal/issues/358#issuecomment-426147844

jamienoss, 2018-10-02:
I'm going to close this, we can then keep any dialogue to #358 where most of the discussion already exists.

closed 2018-10-02 by jamienoss