spacetelescope/hstcal #358: WFC3 inconsistency in CTE processed data between Linux and MacOS using multiple threads

mdlpstsci, 2018-09-21:
There is an inconsistency between Linux and MacOS FLC output datasets which were previously compatible in the pre-HSTDP-2018.3 era (i.e., the same output data products were produced
as determined via "fitsdiff" with zero tolerance).  The datasets were processed with the Astroconda distribution HSTDP-2018.3 (HSTCAL 2.1.0rc2).
 
1) The FLC files processed using multiple threads on Linux and MacOS were found to be incompatible with a tolerance of zero.
 
2) The FLC files processed forcing the use of a single thread on Linux and MacOS were found to be entirely compatible with a tolerance of zero - GOOD.

This issue is related to Issue #357.

mdlpstsci, 2018-09-21:
To capture some email discussion though the regression testing and truth files are not part of this specific issue...

Michele:
I did do a quick test yesterday using your fix (PR #126), and it did not seem to be the culprit. 

Jamie:
So to clarify, the only diff is the multi threaded linux case, right?
 
Which compiler and version are you using?
Have you tried using clang and llvm-openmp on the linux machine (make sure the GCC's openmp library doesn’t exist so that it isn’t picked up instead)?
 
Just to point out that even with the above fix, you will get different results because, in this case, the truth files are the ones that are actually wrong. The other tests just happen to be consistent, but not correct. Whilst this might seem unlikely, it is possible. If you were to ask me what the chance of suppressing this inconsistency by adding stackprotection was, I would say infinitesimal. But then, here we are, it happened.

mdlpstsci, 2018-09-21:
More email...

Michele:
The Linux single thread output matches the MacOS output where the MacOS output is the same regardless of the number of threads.
 
Jamie:
Cool so it is only the multi threaded linux case that is out.
 
Regarding my comment about your quick tests, all I’m saying is that, you should not expect #126 (the fix for issue #61) to necessarily give you the same results as the others and the truth (since those are the same). So if you compared the fixed multi threaded test run against previously run results, they may not be identical and that would be ok. I would merge the fix and then run single and multi threaded results (but never using any previously generated data) and see fi they are the same, they should be. If they’re not, I would then build the fixed code on OSX and run single and multi and compare, just as a sanity check, i.e. it may not be only the linux multi case that is out. One bug might be masking another.
 
 
Another thing to point out is that Anaconda 5.0 deprecated the gcc and libgcc dists. which we are using to build hstcal with 
We should consider the implications of this.
 
Line in recipe: https://github.com/astroconda/astroconda-dev/blob/master/hstcal/meta.yaml#L25
 
Mention of deprecation: https://github.com/conda/conda/issues/7775#issuecomment-422159927
 
"Deprecation of gcc was only implicit in that blog post.” (whatever that means) :https://github.com/conda/conda/issues/7775#issuecomment-422434100
 
Said blog: https://www.anaconda.com/blog/developer-blog/utilizing-the-new-compilers-in-anaconda-distribution-5/ (as pointed to from here: https://github.com/ContinuumIO/anaconda-issues/issues/10081#issuecomment-422545507)

Conda compiler doc page
https://conda.io/docs/user-guide/tasks/build-packages/compiler-tools.html
 
I also noticed when discussing OpenMP with HEALpix that conda’s gcc and libomp are odd (ref: https://github.com/astropy/astropy-healpix/issues/78#issuecomment-418845548). We may not want to be using them for this reason also, they may contain, other oddities.

jamienoss, 2018-09-21:
This may be related to https://github.com/spacetelescope/hstcal/issues/320

jamienoss, 2018-09-22:
More email...

Jamie:
>Another thing to point out is that Anaconda 5.0 deprecated the gcc and libgcc dists. which we are using to build hstcal with 
We should consider the implications of this.
> Line in recipe: https://github.com/astroconda/astroconda-dev/blob/master/hstcal/meta.yaml#L25
>
> Mention of deprecation: https://github.com/conda/conda/issues/7775#issuecomment-422159927
>
> "Deprecation [...]
https://conda.io/docs/user-guide/tasks/build-packages/compiler-tools.html
>
>I also noticed when discussing OpenMP with HEALpix that conda’s gcc and libomp are odd (ref: https://github.com/astropy/astropy-healpix/issues/78#issuecomment-418845548). We may not want to be using them for this reason also, they may contain, other oddities.

jamienoss, 2018-09-22:
FYI: If there is another bug/issue at play here, I wouldn't try to fix it without #126, as we might all be chasing our tails for a while. Once this issue has been resolved and if, for whatever reason, #126 is still not wanted, make this face :feelsgood: and just remove it (don't worry there won't be any zombie dogs or mutant Nazis with gatling guns for hands).

stscicrawford, 2018-09-27:
Working through several different combinations of compiler and code, the problem seems related to the current gcc compiler available at Anaconda here: 
https://anaconda.org/anaconda/gcc/files

If the files are built with an older gcc compiler (for example, gcc version 4.4.7  available on the linux machines) or a newer one (for example, the new [gcc_linux](https://anaconda.org/anaconda/gcc_linux-64/files)), the current HSTCal code seems able to reproduce the 2018.1 results regardless of how or where it is run for at least one representative file.  

As linked above, the Anaconda project as depreciated the building of gcc and now uses OS-specific compilers.   So the build instructions for HSTCal will need to be updated in astroconda.

jamienoss, 2018-09-27:
I thought we covered the symptom expression being caused by the compiler change in https://github.com/spacetelescope/hstcal/issues/320? Maybe not, I can't remember.

Anyhow, sorry to sound like a broken record but have we run the correct comparisons for #126 yet?  The backend of a compiler only has to provide sequential consistency and not introduce any race conditions that aren't already present, basically, it's free to do as it pleases. My point is that the version of ``gcc`` that has been highlighted as "the culprit" may not actually be at fault, it may just be different (which it's allowed to be). #61 is build-unstable.

I'm sorry to keep driving at this point but given the nature of this issue and #61, and assuming that others understand it, I can't understand why anyone would choose to bug hunt without that fix. Without it, anything you find is completely meaningless (now that we know #61 exists) because it may _just_ be a manifestation of #61 and not something else.

To be clear, I'm not saying that this issue _is_ #61, it may not be, just that without #126 you're climbing a scree slope.

stscicrawford, 2018-09-27:
Just to double check, I've check out #126 and these are the comparisons:

With a compiler that reproduces the 2018.1 result: there is 99% difference with the 2018.1 results
With a compiler that produces the 2018.3 result: there is a 99% difference with the 2018.1 results and 99% difference from the currently produced 2018.3 results. 

While #61 may be a problem, it does not create the same results as the 2018.1 which is our current goal.  Changing the compiler used does seem to create results that match 2018.1.

jamienoss, 2018-09-28:
> With a compiler that reproduces the 2018.1 result: there is 99% difference with the 2018.1 results

This is ok, it's expected as we know for a fact that the 2018.1 WFC3 CTE data is wrong.

> With a compiler that produces the 2018.3 result: ... and 99% difference from the currently produced 2018.3 results.

Again ok, and to be expected beacuse the 2018.3 results are _at least_ as wrong as the 2018.1 results.

> With a compiler that produces the 2018.3 result: there is a 99% difference with the 2018.1 results 

The __critical__ question here is which 2018.1 results? The original "target" or that produced from the code that includes #126? If the former,  that's ok and expected. If the latter, this issue is not #61. However, you should still keep #126 whilst trying to track down the other issue.

>it does not create the same results as the 2018.1 which is our current goal. 

This is what I fear. I'm afraid this is a strong indicator that #61 hasn't been understood. This issue _shouldn't_ be about obtaining the 2018.1 results.  This issue is about unwanted data variation and, therefore, the solution should be only about removing this variation. What if the 2018.1 target data is __not__ the basin of attraction? What if you're aiming for the wrong target?

jamienoss, 2018-10-02:
So... this issue ___is___ #61.

 * __Machine:__ INS Linux box (REHL Server release 6.10 (Santiago))
 * __Compiler:__ Anaconda's gcc  4.8.5 (build 7)
 * __Built with OpenMP support__: yes
 * __Tag:__  2.1.0
 * __Data:__ iacr51ohq_raw.fits
 * __Reference files:__ whatever's in ``iref`` on the day of writing this.

## Reproduction
Running ``calwf3.e`` on the above data file, multi-threaded and single-threaded (i.e. using the cmd line option ``-1``, and __NOT__ rebuilding the code without OpenMP support) produces difference that ``fitsdiff`` detects ~95% different pixels in both ``SCI`` image extensions of the CTE corrected FLC calibrated data. This is an indicator that there is a problem that should not exist, and one that is compiler/OpenMP related.

## Bug Hunt
Since the bug is OpenMP related what OpenMP code exists in the ``calwf3`` code base:

 * ``omp parallel for`` [Lines 875 - 902](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L875-L902)
 * ``omp parallel for``  [Lines 920 - 937](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L920-L937)
 *  ``omp critical``  [Lines 933 - 936](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L933-L936)
 * ``omp parallel for``  [Lines 1241 - 1373](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1241-L1373). This is the main CTE correction code.
 * ``omp critical`` [Lines 1359 - ????](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L1359). See issue to be opened (the crit sec should be scoped for crystal clarity).

So, let's comment out each of these OpenMP directives individually and see what happens. Actually, let's just start & finish with the 1st as it happens to be the culprit.

```c
...
//#pragma omp parallel for schedule(dynamic) \
        private(i,j,imid,obs_loc,rsz_loc,dptr)\
        shared(raz, rsz, rnsig,rms,nrms, zadj)
        for(i=0; i<RAZ_COLS; i++){
            imid=i;
            /*RESET TO MIDDLE RAZ_COLS AT ENDPOINTS*/
            if (imid < 1)
                imid=1;
            if (imid == RAZ_COLS-1)
                imid = RAZ_COLS-2;

            /*COPY THE MIDDLE AND NEIGHBORING PIXELS FOR ANALYSIS*/
            for(j=0; j<RAZ_ROWS; j++){
                obs_loc[0][j] = Pix(raz->sci.data,imid-1,j);
                obs_loc[1][j] = Pix(raz->sci.data,imid,j);
                obs_loc[2][j] = Pix(raz->sci.data,imid+1,j);

                rsz_loc[0][j] = Pix(rsz->sci.data,imid-1,j);
                rsz_loc[1][j] = Pix(rsz->sci.data,imid,j);
                rsz_loc[2][j] = Pix(rsz->sci.data,imid+1,j);
            }
            for (j=0; j<RAZ_ROWS; j++){
             if(Pix(raz->dq.data,imid,j)) {
                find_dadj(1+i-imid,j, obs_loc, rsz_loc, rnsig, &dptr); // <----- UH OH
                Pix(zadj.sci.data,i,j) = dptr;
              }
            }
        } /*end the parallel for*/
...
```

Hmmmmm, ``find_dadj()`` is called within that block and happens to be the function containing #61.

What happens if we comment out [lines 875 - 877](https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/wf3cte/wf3cte.c#L875-L877), the ``#pragma omp parallel for``? The resultant FLC data is identical between the multi and single threaded cases _and_ identical to the FLC produced by the 2.1.0 single-threaded run. Therefore, the bug is bound to the code path bound by that for-loop. In addition, given that only this single OpenMP directive was removed, and the others remained, it is significantly less likely that there is a bug within the OpenMP section of the compiler or the runtime library, instead, it is a strong indicator that the bug is within the hstcal code base itself.

Furthermore, cherry-picking #126 (the fix for #61) on top of 2.1.0 and building and running as described above, no difference is found in the resultant FLC data between single- and multi-threaded runs.

It _was_ #61 all along.

mdlpstsci, 2018-10-02:
Woo Hoo - excellent diagnosis.

jamienoss, 2018-10-02:
I guess the next thing to find out is whether this data instability is _just_ due to changes to the code and/or compiler implementation, or is it a function of the number of threads used.

Was this last aspect ever verified, even for whatever compiler you determined suppressed the symptom?

mdlpstsci, 2018-10-02:
Number of threads was not explored (aside from 1 versus not 1).

stscicrawford, 2018-10-02:
Okay, interesting.  Is it the combination of the patch and removing the parallel aspects?  I did run with just the patch, and that didn't seem to fix the problem.   

If so, it would be good to update #126 with both updates and then see if we can confirm that it builds under all environments using the original conda recipe and does reproduce the 2018.1 results.

jamienoss, 2018-10-02:
@stscicrawford 
> Is it the combination of the patch and removing the parallel aspects?

No, I never used the combination of the two. The removal of that parallel section was to demonstrate that the issue is bound to that section of the code.

> If so, it would be good to update #126 with both updates

No, that parallel region should remain.

> I did run with just the patch, and that didn't seem to fix the problem.

You're trying to solve a fictional problem.

> and does reproduce the 2018.1 results.

That is impossible. Please, look over #61 and #126. The 2018.1 results are wrong, they are not scientifically accurate as the code is reading memory from beyond the array, i.e. probably some other instruction or something else on the stack.

jamienoss, 2018-10-03:
Uh Oh! I just found a dead canary.

Same specs as in https://github.com/spacetelescope/hstcal/issues/358#issuecomment-426147844 but with #126 cherry-picked on top of https://github.com/spacetelescope/hstcal/commit/670d8b5a27c611baf1173f54adec862a66375103 (2.1.0).

``iacr510hq_raw.fits`` processed with ``calwf3.e`` and resultant CTE corrected FLC file compared (using ``fitsdiff``) with ``nott:/srv/rtx/hstcal/wf3/onorbit/uvis/ref/iacr510hq_flc.fits`` yeilds (_trouble in paradise_):

 | rtol=1E-n | 1 | 2 | 3 | 4 | 5 | 6 | 7 |
 |---|---|---|---|---|---|---|---|  
 |SCI 1 % pix diff|  0.07 | 0.74 |8.61 |41.07|72.1|94.53|99.18|
 |SCI 2 % pix diff |0.13|1.3|11.06|67.81|95.02|99.2|99.9|

jamienoss, 2018-10-03:
@mdlpstsci provided me with the 2018.1 version of ``iacr510hq_flc.fits``, the image data is identical  to that in rtx as compared in https://github.com/spacetelescope/hstcal/issues/358#issuecomment-426673572. 

The above comparison is thus an accurate comparison to 2018.1 results.

stscicrawford, 2018-10-04:
Here is a summary of some other results for reference:

Machine: INS Linux box (REHL Server release 6.10 (Santiago))
Compiler: Anaconda's gcc 4.8.5 (build 7)
Built with OpenMP support: yes
Tag: 2.2.0a 
Data: iacr51ohq_raw.fits
Reference files: whatever's in iref on the day of writing this.

Compared to 2018.1 iacr510hq_flc
 | rtol=1E-n | 1 | 2 | 3 | 4 | 5 | 6 | 7 |
 |---|---|---|---|---|---|---|---|  
 |SCI 1 % pix diff|  0.16 |1.70 |18.41 | 59.42 | 91.34 | 98.67 | 99.84|
 |SCI 2 % pix diff |0.23 |2.25| 19.07 |78.97 |96.83 |99.49 | 99.94

Machine: INS Linux box (REHL Server release 6.10 (Santiago))
Compiler: gcc version 4.4.7 20120313 (Red Hat 4.4.7-23) (GCC) 
Built with OpenMP support: yes
Tag: 2.2.0a  
Data: iacr51ohq_raw.fits
Reference files: whatever's in iref on the day of writing this.

No difference with 2018.1 iacr510hq_flc

Machine: INS Linux box (REHL Server release 6.10 (Santiago))
Compiler: gcc version 4.4.7 20120313 (Red Hat 4.4.7-23) (GCC) 
Built with OpenMP support: yes
Tag: 2.2.0a + #126 cherry picked
Data: iacr51ohq_raw.fits
Reference files: whatever's in iref on the day of writing this.

 | rtol=1E-n | 1 | 2 | 3 | 4 | 5 | 6 | 7 |
 |---|---|---|---|---|---|---|---|  
 |SCI 1 % pix diff|  0.07 | 0.74 |8.61 |41.07|72.1|94.53|99.18|
 |SCI 2 % pix diff |0.13|1.3|11.06|67.81|95.02|99.2|99.9|

(Exactly the same as reported above using the current conda gcc compiler)

jamienoss, 2018-10-04:
@stscicrawford Your results are indicative that there is not an issue with the compiler, at least not between REHL 6.10 & Anaconda distributed gcc 4.8.5-7. If there were an issue with the compiler, if the original data diff were not a consequence of #61, your second set of results (2.2.0a + #126) should not match mine.

stscicrawford, 2018-10-04:
The other information that could be added is that 'No difference with 2018.1 iacr510hq_flc' were also found using `clang-0.9` and a later version of `gcc` along with the tests that Michele has run with different compilers and architectures.    The bug also didn't show up when creating 2018.1 or earlier versions.

So I would agree that it is not a compiler 'issue', but there is something in the settings of the conda `gcc` that causes the underlying bug to be revealed which doesn't show up when using other compilers or did not previously show up in early builds. 

Nonetheless, I do agree that #61 is the underlying issue.    

So do we have any theories for why the code works under some compilers and not others?  And also, is there anything else in the code that might needs to be potentially looked at that needs to be updated in response to this update to return the same results? Is there any pattern in the pixels most effected by the changes?    Since we have a little bit of time, I would like to do our due diligence to explain the differences as they are significant.     

After doing that, then I'd suggest provide DPAS a version of HSTCal with #126 merged and  running the full WFC3 regression tests again for approval by WFC3 after explaining the situation to them.   If they are happy with the results, then we can have an updated release and quickfix for the pipeline.

jamienoss, 2018-10-04:
@stscicrawford 
>So do we have any theories for why the code works under some compilers and not others? 

Nothing concrete, only that they are allowed to be different, and therefore whatever is being read on the stack (that shouldn't be) is different. We could print these out and see what the actual values are, we _could_ also look at the assembly and see what instruction (or other piece of data) is being read.

I think the next thing to do is explore the different optimization levels, i.e. none, -O2, & -O3. I think these might be the simplest control over what the compiler does.

>hat causes the underlying bug to be revealed which doesn't show up when using other compilers or did not previously show up in early builds

I'm not sure if this is actually true, especially when you consider the possibility of #61 being unstable to compiler optimizations. Those other builds were tested with _only_ -O2 (I think), so they could have been different.

Another avenue is to back out any of the code changes in ``wfc3cte.c`` and see if any changes in the code could have caused the compiler to reorder stuff. When I worked on the ACS CTE rewrite, #126 was the only way I could get the results to match (my original rewrite was actually of wfc3cte and tested only with wfc3 data until it matched, then I moved onto ACS).

jamienoss, 2018-10-04:
For example, it might be good to revert #93 and re-run some quick tests, just as a sanity check.

stscicrawford, 2018-10-09:
Just as some further information, here are the ratio and difference images between 2018.1 and 2018.3+#126:

Output image:
[image]

Difference image -- The difference are at a level ranging from 0.003 on the left side, and 1e-5 on the right side: 
[image]

Ratio image -- Most of the results are around <1.001.  The difference images are fairly consistent level, and so variation in here is mainly driven by the 
[image]

For reference, the readnoise in the image is ~3 electrons.

stscicrawford, 2019-02-02:
Closed after review by WFC3 team that was happy with the results from #366

closed 2019-02-02 by stscicrawford