spacetelescope/hstcal #359: WAF incompatible with Python 3.7

jamienoss, 2018-10-01:
I get the following when trying to configure hstcal:

```
raceback (most recent call last):
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Node.py", line 295, in ant_iter
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Scripting.py", line 120, in waf_entry_point
    run_commands()
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Scripting.py", line 177, in run_commands
    parse_options()
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Scripting.py", line 150, in parse_options
    Context.create_context('options').execute()
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Options.py", line 149, in execute
    super(OptionsContext,self).execute()
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Context.py", line 88, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Context.py", line 129, in recurse
    user_function(self)
  File "/home/jnoss/dev/hstcal/wscript", line 48, in options
    opt.load('compiler_c')
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Context.py", line 85, in load
    fun(self)
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Tools/compiler_c.py", line 40, in options
    opt.load_special_tools('c_*.py',ban=['c_dumbpreproc.py'])
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Context.py", line 319, in load_special_tools
    lst=self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Node.py", line 344, in ant_glob
    ret=[x for x in self.ant_iter(accept=accept,pats=[to_pat(incl),to_pat(excl)],maxdepth=kw.get('maxdepth',25),dir=dir,src=src,remove=kw.get('remove',True))]
  File "/home/jnoss/dev/hstcal/.waf3-1.9.9-9d160c52ab497577bf419cd4334f62b4/waflib/Node.py", line 344, in <listcomp>
    ret=[x for x in self.ant_iter(accept=accept,pats=[to_pat(incl),to_pat(excl)],maxdepth=kw.get('maxdepth',25),dir=dir,src=src,remove=kw.get('remove',True))]
RuntimeError: generator raised StopIteration
```

A quick Google yields a similar issue https://github.com/pypa/setuptools/issues/1285 which states [here](https://github.com/pypa/setuptools/issues/1285#issuecomment-370560419)

> The issue is an intentional regression introduced in Python 3.7 as part of [PEP 479](https://www.python.org/dev/peps/pep-0479).

I had a quick search over the [WAF issues](https://gitlab.com/ita1024/waf/issues) and couldn't see anything related, this may have not been fixed.

Anyone else getting deja vu with this issue?

jamienoss, 2018-10-01:
FYI @rendinam @jhunkeler

pllim, 2018-10-01:
Do we have a pressing need to support HST pipeline in Python 3.7? Anyway, perhaps @stscicrawford can establish a timeline for this.

jamienoss, 2018-10-01:
https://conda.io/miniconda.html default ships with either 2.7 or 3.7. That's prob the only "need", simple builds for newbie developers (what ever they are 😉).

pllim, 2018-10-02:
Well, regardless, it is probably worth opening an issue upstream at https://gitlab.com/ita1024/waf/issues

pllim, 2018-10-02:
And to bundle a new version of `waf`, hopefully the steps at #15 still works.

mdlpstsci, 2022-01-12:
HSTCAL is currently using waf-2.0.19, and it works fine with Python 3.8 which is the version of Python now being used for the builds.
Resolved.

closed 2022-01-12 by mdlpstsci