spacetelescope/hstcal #373: ACSREJ: Refactoring the treatment of pixels marked in the BPIXTAB

nmiles2718, 2018-10-30:
After the discovery of [flecks on the ACS/WFC CCD](http://www.stsci.edu/hst/acs/documents/isrs/isr1803.pdf), the ACS Team created a new bad pixel table (BPIXTAB) to flag pixels contaminated by the fleck during processing. In the midst of creating the new BPIXTAB, @samanthalh discovered that the values for the flags in the BPIXTAB were incorrect (they were supposed to have a value of 4, but they were marked with 1024). 

The information stored in the BPIXTAB is propagated to the DQ arrays of the input file in `acsccd` during the doDQI step. This means they are present before the call to ACSREJ. After updating the BPIXTAB to use the correct value of 4, @jryon noticed that the imprint of the flecks began appearing in the SCI and ERR extensions of superbias files that are created by stacking a month of bias frames together. The flecks are caused by particulates on the surface of the CCD and therefore only affect data where light (from external sources, the post-flash LED, or lamps) is incident on the surface of the CCD. Hence, they do not and should not appear in the SCI and ERR extensions of individual bias frames, which contain only signal from the detector and electronics. This suggests that ACSREJ has internal logic to treat pixels flagged with DQ=4 differently in the SCI and ERR arrays, which went unnoticed because the BPIXTAB was previously using the incorrect values.

An example of the effects can be seen in the figure below:
[image]

The desired behavior would be for ACSREJ to propagate the BPIXTAB values from the individual DQ arrays to the final CRJ product, while ignoring them in the SCI and ERR arrays during the combination process (i.e. treat them as normal pixels).

nmiles2718, 2018-11-08:
FYI:
For the pixels marked in the BPIXTAB, the values are replaced by the sky value that is computed by `ASCREJ`. If you stack a set of images with `SKYSUB=‘none’`, then all the pixels marked in the BPIXTAB will have a value of 0. If you set it to `SKYSUB=‘mode’`, then they will have whatever value is computed by ACSREJ. Given this, I would guess that `ACSREJ` sniffs the DQ during some initial step and gathers information about pixels marked in the BPIXTAB before it does anything else. Hopefully this helps to narrow things down some, but if not, it at least provides us with an easy way to test the new behavior after updates are made.

jryon, 2018-11-28:
Some discussion with Norman has led us to change the desired behavior for this issue with ACSREJ. 

For biases and darks only, we'd like ACSREJ to behave as described by @nmiles2718 above, i.e., "propagate the BPIXTAB values from the individual DQ arrays to the final CRJ product, while ignoring them in the SCI and ERR arrays during the combination process (i.e. treat them as normal pixels)."

For all other data, ACSREJ should behave as it currently does, i.e., do not include pixels flagged with DQ flag = 4 when stacking images. In other words, the SCI and ERR arrays of the final, combined image should not include the contribution from pixels flagged with DQ flag = 4. The DQ array of the final, combined image should not contain DQ flag = 4. Effectively, CR rejection and bad pixel rejection from the final stack are should behave the same in this case.

jryon, 2019-01-10:
This issue affects the creation of the ACS bias and dark reference files, and is therefore fairly urgent. Several currently active reference files have pixels incorrectly masked by the BPIXTAB, and any new reference files create will be affected as well. We cannot remake them until this issue is resolved.

pllim, 2019-01-10:
cc @stscicrawford and @mdlpstsci

mdlpstsci, 2019-01-11:
Jenna, thanks for adding the criticality commentary per our meeting on Wednesday.

mdlpstsci, 2019-02-19:
@nmiles2718 @jryon FYI - I have time to investigate this issue this week. Please consider this issue "In Progress".

mdlpstsci, 2019-02-19:
@nmiles2718 @jryon Do one of you have some good datasets (sci, bias, dark) that can be used for testing?

nmiles2718, 2019-02-19:
@mdlpstsci I'll make a good testing dataset for you and place it on my central store at in a directory called `ACSREJ_BPIXTAB` with the following structure:
- `ACSREJ_BPIXTAB/testing_datasets/`
   - `dark/`: a list of RAW dark frames
   - `bias/`: a list of RAW bias frames
   - `external/`: a list of RAW external science images from a CR-SPLIT observation

I'll send you an email once I've finished 👍

mdlpstsci, 2019-10-23:
Fixed in PR #402.

closed 2019-10-23 by mdlpstsci