spacetelescope/hstcal #401: CALSTIS: Incorrect x1d/x2d extension applied instead of sx1/sx2 in some cases

jkcarlberg, 2019-02-27:
The extensions applied by calstis to calibrated data products distinguish between data products that came from individual datasets (_x1d and _x2d files) and data that came from summed observations (e.g., CR-SPLITs). The latter should have extensions _sx1 and _sx2.  However, after using the standalone ocrreject task for custom cosmic ray rejection, the final data products are erroneously given the _x1d and _x2d extensions.   The problem appears to be in calstis0.c which checks whether the keyword CRCORR is set to PERFORM, but should also be checking whether it is set to COMPLETE.

mdlpstsci, 2019-02-27:
@SaOgaz @stscirij - Just making sure you see this.

pllim, 2025-06-10:
@stscirij is this still a problem?