spacetelescope/hstcal #405: Upstream release of dependency: cfitsio

rendinam, 2019-05-24:
This is a message from an automated system that monitors `cfitsio` releases.

Version 3.47 - May 2019

  - Added set of drivers for performing ftps file transfers.

  - Tile sizes for compression may now be specified for any pair of
    axes, where previously 2D tiles where limited to just X and y.
    
  - Fix to ffgsky and ffgkls functions for case of keyword with long
    string values where the final CONTINUE statement ended with '&'.
    If the final CONTINUE also contained a comment, it was being 
    repeated twice when passed back through the 'comm' argument.
      
  - Fix made to ffedit_columns() for case of multiple col filters
    containing wildcards.  Only the first filter was being searched.
                     
  - fits_copy_rows (ffcprw) can now handle 'P'-type variable-length 
    columns.
    
  - Fix made to an obscure case in fits_modify_vector_len, where a 
    wrongly issued EOF error may occur.
                   
  - Added internal fffvcl() function.
                   

(For complete changelog information, consult the package changelog file in cfitsio.../doc/changes.txt)

  **NOTE: This release introduces a SONAME change from 7 to 8.**

pllim, 2019-09-16:
Astropy is doing it over at astropy/astropy#9233 . FYI.

mdlpstsci, 2022-01-11:
There is a newer, major release of CFITSIO (4.0.0 - July 2021) against which HSTCAL should be tested.

closed 2022-01-11 by mdlpstsci