spacetelescope/hstcal #406: Unable to run calwf3 due to problem with libcfitsio

bhilbert4, 2019-05-30:
I just created a new environment following the instructions in the repo README, but when I try to run calwf3, I'm getting an error related to libcfitsio.6.dylib. Do I need to go find and download this library from somewhere?

```
> conda create -n calwf3_env python=3 stsci-hst
> conda activate calwf3_env
> calwf3.e idfs11doq_raw.fits 
dyld: Library not loaded: @rpath/libcfitsio.6.dylib
  Referenced from: /Users/hilbert/anaconda3/envs/calwf3_env/bin/calwf3.e
  Reason: image not found
Abort trap: 6
```

rendinam, 2019-05-30:
Could you please provide the output of the command `conda list` from a command prompt within the activated `calwf3_env` environment?

bhilbert4, 2019-05-30:
```
# packages in environment at /Users/hilbert/anaconda3/envs/calwf3_env:
#
# Name                    Version                   Build  Channel
acstools                  3.0.0                    py37_0    http://ssb.stsci.edu/astroconda
asdf                      2.3.1                      py_0    conda-forge
asn1crypto                0.24.0                py37_1003    conda-forge
astropy                   3.1.2            py37h1de35cc_0    conda-forge
atomicwrites              1.3.0                      py_0    conda-forge
attrs                     19.1.0                     py_0    conda-forge
beautifulsoup4            4.7.1                 py37_1001    conda-forge
bzip2                     1.0.6             h1de35cc_1002    conda-forge
ca-certificates           2019.3.9             hecc5488_0    conda-forge
calcos                    3.3.5                    py37_0    http://ssb.stsci.edu/astroconda
certifi                   2019.3.9                 py37_0    conda-forge
cffi                      1.12.3           py37hccf1714_0    conda-forge
cfitsio                   3.430             h50cca31_1002    conda-forge
chardet                   3.0.4                 py37_1003    conda-forge
cloudpickle               1.0.0                      py_0    conda-forge
costools                  1.2.3                    py37_0    http://ssb.stsci.edu/astroconda
crds                      7.3.3                    py37_0    http://ssb.stsci.edu/astroconda
cryptography              2.6.1            py37h212c5bf_0    conda-forge
curl                      7.64.1               h22ea746_0    conda-forge
cycler                    0.10.0                     py_1    conda-forge
cython                    0.29.9           py37h6de7cb9_0    conda-forge
cytoolz                   0.9.0.1          py37h470a237_1    conda-forge
d2to1                     0.2.12.post1               py_0    conda-forge
dask-core                 1.2.2                      py_0    conda-forge
decorator                 4.4.0                      py_0    conda-forge
drizzlepac                2.2.6                    pypi_0    pypi
filelock                  3.0.10                     py_0    conda-forge
fitsblender               0.3.2                    py37_1    http://ssb.stsci.edu/astroconda
fitsverify                4.18                          5    http://ssb.stsci.edu/astroconda
freetype                  2.10.0               h24853df_0    conda-forge
hstcal                    2.2.0                         0    http://ssb.stsci.edu/astroconda
icu                       58.2              h0a44026_1000    conda-forge
idna                      2.8                   py37_1000    conda-forge
imageio                   2.5.0                    py37_0    conda-forge
importlib_metadata        0.17                     py37_0    conda-forge
jpeg                      9c                h1de35cc_1001    conda-forge
jsonschema                3.0.1                    py37_0    conda-forge
kiwisolver                1.1.0            py37h770b8ee_0    conda-forge
krb5                      1.16.3            hcfa6398_1001    conda-forge
libblas                   3.8.0               10_openblas    conda-forge
libcblas                  3.8.0               10_openblas    conda-forge
libcurl                   7.64.1               h16faf7d_0    conda-forge
libcxx                    8.0.0                         2    conda-forge
libcxxabi                 8.0.0                         2    conda-forge
libedit                   3.1.20170329      hcfe32e1_1001    conda-forge
libffi                    3.2.1             h6de7cb9_1006    conda-forge
libgcc                    4.8.5               hdbeacc1_10    conda-forge
libgfortran               3.0.1                         0    conda-forge
libiconv                  1.15              h01d97ff_1005    conda-forge
liblapack                 3.8.0               10_openblas    conda-forge
libpng                    1.6.37               h2573ce8_0    conda-forge
libssh2                   1.8.2                hcdc9a53_2    conda-forge
libtiff                   4.0.10            h79f4b77_1001    conda-forge
libxml2                   2.9.9                hd80cff7_0    conda-forge
libxslt                   1.1.32            h33a18ac_1002    conda-forge
lxml                      4.3.3            py37h08abf6f_0    conda-forge
matplotlib                3.1.0                    py37_1    conda-forge
matplotlib-base           3.1.0            py37h3a684a6_1    conda-forge
more-itertools            4.3.0                 py37_1000    conda-forge
ncurses                   6.1               h0a44026_1002    conda-forge
networkx                  2.3                        py_0    conda-forge
nictools                  1.1.4                    py37_0    http://ssb.stsci.edu/astroconda
nose                      1.3.7                 py37_1002    conda-forge
numpy                     1.16.4           py37h6b0580a_0    conda-forge
olefile                   0.46                       py_0    conda-forge
openblas                  0.3.6                hd44dcd8_2    conda-forge
[... 35 lines cut ...]
stregion                  1.1.4                    py37_0    http://ssb.stsci.edu/astroconda
stsci-hst                 3.1.1                    py37_0    http://ssb.stsci.edu/astroconda
stsci.convolve            2.2.2                    py37_1    http://ssb.stsci.edu/astroconda
[... cut ...]
wfc3tools                 1.3.5                    py37_0    http://ssb.stsci.edu/astroconda
wfpc2tools                1.0.4                    py37_0    http://ssb.stsci.edu/astroconda
wheel                     0.33.4                   py37_0    conda-forge
xz                        5.2.4             h1de35cc_1001    conda-forge
yaml                      0.1.7             h1de35cc_1001    conda-forge
zipp                      0.5.0                      py_0    conda-forge
zlib                      1.2.11            h1de35cc_1004    conda-forge
```

rendinam, 2019-05-30:
The `conda-forge` is likely causing problems here. We do not recommend its use in this context.
The command `conda config --remove channels conda-forge` should disable it and allow you to remove and create the environment again.
`$ conda config --remove channels conda-forge`
`$ conda remove -n calwf3_env --all`
`$ conda create -n calwf3_env python=3 stsci-hst`

bhilbert4, 2019-05-30:
Thanks! This has solved the problem. I'll keep `conda-forge` at a lower priority than the `astroconda` channel from now on.

closed 2019-05-30 by pllim