spacetelescope/hstcal #435: Change CALACS to support time-dependent sensitivity for SBC

rjavila, 2019-09-25:
I am in the process of implementing the time-dependent sensitivity for the SBC. I’m now testing the IMPHTTAB and I’m finding that the PHOTFLAM values are not being updated correctly. CALACS is not properly updating the PHOTMODE keyword (EXT=1) of FLT images. Currently it looks like this for SBC:

PHOTMODE= 'ACS SBC F115LP'     / obser  

In WFC images the PHOTMODE includes an MJD, and that’s how CALACS knows to apply TDS. It looks like the culprit is the code block at line 103 in the photmode.c file.

mdlpstsci, 2019-09-25:
The modification is to remove the "if" statements surrounding the update of the PHOTMODE keyword which currently only allow the update to HRC and WFC.  @rjavila do you have a IMPHTTAB file I can use for testing as CALACS fails 

Found parameterized variable 1.
NUMPAR=1, N=1
Allocated 1 parnames
Adding parameter mjd#56392.6597 as parnames[0]
==>ERROR: Matching row not found in IMPHTTAB `acs,sbc,f150lp,mjd#'.
==>ERROR: OBSMODE acs,sbc,f150lp,mjd#
ERROR:    Error return from GetPhotTab.

mdlpstsci, 2019-09-30:
The hstdp-2019.4 build containing this fix has been generated:
https://github.com/astroconda/astroconda-releases/tree/master/hstdp/2019.4

HSTCAL Release Notes:
https://github.com/spacetelescope/hstcal/releases

mdlpstsci, 2019-10-23:
Fixed in PR #436.

closed 2019-10-23 by mdlpstsci