spacetelescope/hstcal #452: WFC3 CTE new correction algorithm

heatherkurtz, 2019-10-23:
Jay has created an update the the WFC3 CTE correction. The C code has been tested and is ready to be implemented into the pipeline.

CTE correction algorithm has a new pin date and new method of handling the background.

heatherkurtz, 2019-10-23:
This update will be accompanied by a new reference file.

mdlpstsci, 2020-03-03:
After further testing of the algorithm by Jay, he decided to add some additional code.  Jay and I met on 24 February, and he then followed up with an email the next day summarizing our discussion.  Here is the email.
-------
This is the follow-up on our conversation yesterday.  I'll re-cap our conversation, so that Sylvia and Heather will know what was discussed as the plan forward.  

I'm hoping it won't take too much effort to implement the changes below into the "sandbox" you've created for us, so that we can test it and finally get it out to market.

There were two aspects of what we discussed yesterday:
(1) the need to "sense" the noise in the image so that the algorithm wouldn't over- or under-correct for readnoise amplification; and
(2) the need to adjust the CTE correction algorithm for subarrays; the previous algorithm didn't have to treat subarrays differently, but the new one does

The plan for (1) is that if the PCTERNOI keyword in the _raw file is set to something other than zero, then it uses that value of PCTERNOI in the reduction (not the one from the reference file) to run the CTE correction.  

However, if PCTERNOI *is* set to zero to zero, which it is in the RAW images I can see  (it's in the CTE CORRECTION PARAMETERS section on the third "page" of the zeroth header), then the algorithm will adopt the PCTERNOI value returned by the attached subroutine --- float find_raz2rnoival() --- and the algorithm will use this new value in the CTE reduction.  After processing, the PCTERNOI keyword in the flc image will be set to the used value.    The find_raz2rnoival() subroutine returns the PCTERNOI value, which is clipped to be between 3.75 and 9.75.  For my diagnostics, the function also determines an unclipped value for it and a "background" value for the pixels used.  Feel free to remove these passed-by-reference parameters.  I tested the subarray and it appears to give the same results as my stand-alone FORTRAN routines.  Of course I'll re-test this when the sandbox is put together again.

This routine automatically "senses" which pixels in the 8412x2070 raz-format image that is passed to it are in a non-blank region of the detector, so that it can properly deal with subarrays.

The plan for (2) concerns the need for the CTE algorithm to sense which pixels are non-blank.  I'm using the same procedure as above.    Since I wasn't able to test the new version of wf3cte (the one that only operates on non-blank pixels) on my machine on account of the gcc libraries that got lost during my machine update, I couldn't completely check my new routine, but the modifications I made to wf3cte are the same as what I did in the find_raz2rnoival() subroutine.  You can run:

diff -b wf3cte.c wf3cte.c.2020.02.24

To see what changes there are.  Basically, I define two new variables "j1" and "j2" for the column being analyzed and use them to bracket the non-blank regions of the column.

closed 2021-03-24 by mdlpstsci