spacetelescope/hstcal #453: ACS Saturation Map

tddesjardins, 2019-10-28:
The ACS Branch would like to update `calacs` to do full-well saturation flagging (currently part of DQICORR) using an image rather than the current single value in CCDTAB. The CCDTAB uses a single value of full-well saturation for data quality flagging for ALL amplifiers, whereas studies have shown that the full-well spatially varies across each amplifier region. Work on generating these maps is ongoing, and currently there are some open questions on the exact implementation. There are two possible scenarios being discussed:

1. The image (which we'll call "SATUFILE" for the moment) would be checked against during DQICORR. The image would be in units of DN and have two extensions of sizes 4144 x 2068 for WFC and a single 1062 x 1044 extension for HRC. The check would be controlled by the DQICORR header keyword. No check would be made for SBC.

2. Similar to 1 above, but the check would happen at the end of BIASCORR. The image would be in units of electrons and would have calibrated image dimensions after overscan trimming has occurred. 

The ACS Branch will work with CRDS/ReDCaT on implementing the new reference file.

**Aiming to implement this before end of FY2020 Q2 (March 2020) if possible.**

tddesjardins, 2019-10-28:
Also as a note, subarray images will need to use the same cutout idea as what occurs for the 2-D bias correction.

mdlpstsci, 2020-04-08:
I send the interested parties on ACS team an email alerting them to my putting this issue and/or issue #460 into the queue.  Here is the commentary from Tyler on this ticket:

> CC’ing Meaghan and Norman to keep them in the loop, and Yotam since he’s involved in #453 (saturation maps).
> 
> For the saturation maps (#453), if Yotam has the actual map files ready, we can provide you with one for development purposes while we work out how to get them to work with CRDS. Meaghan will need to work with Ed Slavich to define a new reference file type and all of the information that CRDS needs to select it. I think this will also involve some database work to track the name of the reference file for each dataset, so we can ping David Wolfe about that.
>  
> Yotam: Did we ever settle on if you were going to make a map for HRC as well, or are we just going to deliver a dummy file for HRC so the pipeline doesn’t crash? If I recall correctly, we said the correction would be in DN, so it can happen during DQICORR, right?
>

mdlpstsci, 2020-04-15:
Via email from the team on 13 April 2020
Yotam

> I do indeed have the saturation map file for WFC ready. A “development version” of it can be found here: /user/ycohen/saturation_maps/acs_wfc_saturation_map.fits. (let me know if you can’t access that file). At the moment there is no header info in it, so I will briefly explain here, and feel free to slack me or reply to this if you have any questions about it.
>  
> There are two image extensions, the first one corresponding to SCI,1 and the second one corresponding to SCI,2 in full-frame, overscan-trimmed WFC images. The units are electrons. The saturation flagging using this map should be applied to the blv_tmp image immediately after acsccd, or to a temporary image somewhere in between the acsccd steps, but it must come after the bias subtraction and conversion to electrons using the gain. This is in contrast to the current saturation flagging, which if I recall correctly, happens earlier in acsccd. The flagging procedure is straightforward: if the intensity of a given pixel in the blv_tmp is greater than the value of the corresponding pixel in the saturation map, then that pixel should be flagged as saturated. I’m not sure how this works in C, but in python I’m able to do this in one line of code in which I just do something like “saturated_pixels = science_frame > saturation_map” where the output “saturated_pixels” array is just a Boolean array with a value of True for the saturated pixels and False elsewhere.
>  
> As for the HRC, I don’t have any plans at the moment to make the saturation maps for it, but I can certainly look into that if we decide we want to do that.

Norm

> So do I infer correctly that the saturation now needs to 
> happen in BIASCORR rather than DQICORR?  (i.e., as
> immediately as possible after superbias subtraction 
> and gain-conversion)

Tyler

> That’s what it sounds like from Yotam’s description. If the units are electrons and the image is overscan trimmed, it must happen after BLEVCORR/BIASCORR.
>  
> As for HRC Yotam, that’s fine, we can just deliver an image with the right format but a single value that’s just the same as whatever we use now. That way the pipeline acts consistently for both CCDs.

mdlpstsci, 2020-04-17:
On 17 April Meaghan wrote: In regards to the full well saturation update, after checking with Tyler and Yotam, it appears that one new keyword (maybe SATUFILE?) will be required. This will indicate the name of the reference file that was used in case the file ever changes and users will be able to tell which file was used for their data set. Otherwise, no new switches, because it will just use one or two of the existing switches to decide if it needs to be done.

pllim, 2020-04-17:
Fun Fact: "Satu" means "one" in Malay!

mdlpstsci, 2020-04-17:
It is "isa" in Tagalog, but I only speak Bostonian English.

mdlpstsci, 2020-05-08:
This issue is mapped to Jira HLA-340.

mdlpstsci, 2020-05-18:
@yotam-stsci @tddesjardins @mmcdonald123 

In order for me to test the changes I am making in CALACS, I need
to have a reference file which is a bit closer to the true format
than this preliminary file.  I have been using it thus far, and I
have been able to accomplish quite a bit, but...

You will need to confer with ACS team, but I am sure there are
requirements for reference files (ICS-47???) in terms of format
and necessary keywords.  Here are some items which need to be
addressed:

New Full-well Saturation Image Reference File
---------------------------------------------
1) Primary header needs the following keywords and/or value:
   NEXTEND = 6
   PEDIGREE
   DESCRIP
   USEAFTER

2) To match with the current input reference files, this image should
   have 6 extensions (SCI, ERR, and DQ) and EXTVER 1 and 2.
   The EXTNAME and EXTVER keywords should be present in the
   image extensions.

3) The first Imset (SCI/ERR/DQ combo) should be for Chip 2.
   Programmatically this does not matter, but if a human looks in
   the files, we do not want them to be confused.  It looks like
   the ACS reference file order of the Imsets matches that of the
   science data - Chip 2 is the first Imset in the file.

4) The SCI extension needs:
   CCDCHIP
   LTM1_1, LTM2_2, LTV1, LTV2

This is what I have found thus far.  THANKS.

tddesjardins, 2020-05-19:
Thanks for the feedback, @mdlpstsci! Maybe @yotam-stsci and I can have a talk this week about what all this means and how to get it into shape for you. I'll be in touch with him about setting that up (and include Meaghan as well).

mdlpstsci, 2020-06-01:
From Yotam on 28 May 2020:
I’ve placed an updated version of the file here: /user/ycohen/saturation_maps/acs_wfc_saturation_reffile_postsm4.fits
 
If you get a chance to look at it and/or try it out, and it still needs some fixes, please let me know what they are and I will make the changes when I come back from vacation after next week. After we iron out all the fixes, I’ll clone the file for the other USEAFTER date ranges and GAIN levels.

mdlpstsci, 2020-11-30:
@tddesjardins @yotam-stsci 
I would like to verify one item and ask about another.

Verify
(1) Do the BIASCORR and BLEVCORR need to have the values of COMPLETE (in memory) in order to apply the new saturation image?  Please ignore any issues regarding "trim" when you respond to this question as the data is not actually trimmed until the tmp_blv file is written to disk.  In memory, the data is still full-size, unless it is a subarray.  I will also note the data is converted to electrons in a step which is independent of bias and blev.

Question
(2) As noted above, the tmp_blv file is not actually trimmed until the file is physically written to disk.  For several reasons, particularly working code reuse, I intend to embed the saturation image into a full-frame in-memory array where the only pixels impacted by the new saturation image will be those which are in the "trimmed" region.  I am happy to do this, but I wondered if you would want to embed the current saturation image into a full-frame so the calibration file is similar to the other files (e.g., bias image).  Of course, this would require changing LTV1.

Please let me know.

mdlpstsci, 2020-11-30:
@tddesjardins @yotam-stsci 
I should have mentioned that months and months ago, I had assumed bias and blev had to happen.  As such, the saturation image is only read if these steps are going to be performed.  The BIASCORR is used as the trigger for applying the saturation image as there is no "SATUCORR".  If necessary, this can change.

mdlpstsci, 2020-11-30:
@tddesjardins @yotam-stsci 
HI Guys,
Back to ACS work...I just looked at a Bias reference image, and it is full frame with the good bias information embedded in the larger image.   In order to keep the pipeline consistent, I suggest the saturation image be formatted in the same manner as the bias image.  It makes sense from the "deriving a new reference file perspective" the current saturation image is trimmed.  However, from the pipeline perspective, it would be better to embed the useful saturation data in a full frame.  Sorry to write so many messages, but I am just getting back to this work and still wrapping my head around it.

tddesjardins, 2020-11-30:
That's fine, Michele. I think either way is fine, as long as we know which way to do it. But I'm going to bring @mmcdonald123 into this to keep her in the loop as she should be making the calls about pipeline changes from now on. I'm on my way out of the ACS door shortly.

mdlpstsci, 2020-12-01:
@tddesjardins @yotam-stsci @mmcdonald123 
OK.  I think the new saturation image should be formatted in the same manner as the bias calibration file.  By this I mean the current and trimmed 2D saturation data (4096x2048) should be embedded in an image of zeros (4144x2068) and then written out as the new saturation calibration file.  This will keep the pipeline consistent in its methodology and accomplish the task at hand.  Any and all header keywords will need to be updated appropriately.  Please let me know as soon as you have a new file.

mdlpstsci, 2020-12-02:
@yotam-stsci @mmcdonald123 @tddesjardins 

SUMMARY of the items or questions that I have regarding this issue.

1) I am expecting an updated saturation mask from you for ACS WFC data which is
   "full-frame" (i.e., the current saturation data is embedded inside an
   image of zeros which is full-frame).  I mocked up my own version of this
   file so I am continuing with my work.

2) As previously noted by Tyler, you should plan on having a corresponding
   saturation calibration file for ACS HRC with values set appropriately. In
   this way the pipeline does not care which CCD is being processed.

3) In the current public version of the code, pixels which satisfy the
   following

       (value > full-well saturation) OR (value >= A-to-D saturation)

   are flagged with 256.  Note that pixels with

       value >= A-to-D_saturation

   are also flagged with 2048.

   Do you still want this logic?

       (value > full-well saturation) OR (value >= A-to-D saturation)

   or should it just be

       value > full-well saturation

4) As previously noted in Git Issue #453, the application of the SATUFILE
   is triggered by the BIASCORR.  My question here is do the BIAS and BLEV
   corrections need to have been performed in order then to apply the
   SATUFILE information?  I can modify this, but for some reason when
   I originally set this up early this year I had it triggered in this
   manner.  Just so you know the order of correction steps in the pipeline
   are as follows below.  I can move the application of the saturation image
   as you prefer.
     
   BIASCORR
   Conversion to electrons
   BLEVCORR
   Saturation mask application
   NOISECORR
   SINKCORR
   Data written out in trimmed format

5) Does the correction apply to all subarrays or only the "supported
   subarrays"?  I have assumed it applies to all subarrays as the
   "supported subarray" concept was specific for the bias shift correction.
   The answer here somewhat depends upon your answer to Question 4.

yotam-stsci, 2020-12-02:
@mdlpstsci 

> 1. I am expecting an updated saturation mask from you for ACS WFC data which is
>    "full-frame" (i.e., the current saturation data is embedded inside an
>    image of zeros which is full-frame).  I mocked up my own version of this
>    file so I am continuing with my work.

I don't have any saturation information outside the 4096x2048 "trimmed" region. I'm not sure if padding the rest of the image with zeros is exactly what we want though? If you do that, do the pixels outside the trimmed region all get flagged as saturated? If so, we probably wouldn't want that. Ideally I think we'd just want to pixels to be ignored for the saturation flagging step. Otherwise, maybe we can pad the trimmed region with infs or some extremely high value. What are your thoughts on this?

> 2. As previously noted by Tyler, you should plan on having a corresponding
>    saturation calibration file for ACS HRC with values set appropriately. In
>    this way the pipeline does not care which CCD is being processed.

Got it, will generate that ASAP.

> 3. In the current public version of the code, pixels which satisfy the
>    following
>    ```
>    (value > full-well saturation) OR (value >= A-to-D saturation)
>    ```
>
>
>    are flagged with 256. [...]

I'm going to double check with Norman on this one and get back to you.

> 4. As previously noted in Git Issue #453, the application of the SATUFILE
>    is triggered by the BIASCORR.  My question here is do the BIAS and BLEV
>    corrections need to have been performed in [...]

With the way that the saturation mask is currently constructed, the science image has to be converted to electrons and bias-subtracted before the saturation flagging is applied, yes.

> 5. Does the correction apply to all subarrays or only the "supported
>    subarrays"?  I have assumed it applies to all subarrays as the
>    "supported subarray" concept was specific for the bias shift correction.
>    The answer here somewhat depends upon your answer to Question 4.

I'm not entirely sure on this one. I'll check with Norman on this as well.

I'll have more info for you tomorrow.

mdlpstsci, 2020-12-02:
Glad to have your responses to my comment above and look forward to all the answers!  As for Item 1...the area outside of the "trimmed" region will be ignored during the application of the DQ flag.  The reason I am asking to embed the saturation image in a full-frame image is at least so the current pipeline support utilities can continue to function as they do now.  In this way I can minimize the different parts of the code which I have to modify (and potentially introduce bugs).  The new routine written to handle the saturation file is the customized portion and "knows" to ignore the overscan regions.  I just suggested zero as the pad value, but you could use a very large value.  I will ignore these regions to avoid processing a lot of extra data unnecessarily, particularly in the case of subarrays.

Also, do you have an image I can use for testing where there will be pixels flagged with the full-well saturation value?  I created a full-frame mocked up saturation image and force some pixels to be greater than the saturation image value, so I can test my code.  However, I would like to ensure the code works in the real-world case.  Thanks.

yotam-stsci, 2020-12-03:
@mdlpstsci 

For item 1: Thanks for elaborating that point. I think a pad value of zero will actually be fine given that pixels outside the trimmed region will not be in the final products anyway. You said you already hacked up a version of the full-frame saturation mask with that padding, correct? Do you still need me to make a separate copy?

> Also, do you have an image I can use for testing where there will be pixels flagged with the full-well saturation value? I created a full-frame mocked up saturation image and force some pixels to be [...]

Sure, I've put an image in central store: `/user/ycohen/for_others/michele_dlp/jdg301byq_raw.fits`

For item 3: I spoke with Norman and it seems that we are actually going to want to change that logic to the way you proposed. Such that only full-well saturation sets flag 256, and only A-to-D saturation sets flag 2048. I imagine this will require a reprocessing of all the archival WFC data, right? I will also get in contact with the relevant people to update the language in the instrument and data handbooks regarding these flags.

For item 5: I think we're just going to just give subarrays the same treatment as the full-frame images. The program should just use the section of the saturation mask corresponding to the subarray. If the image was not bias-subtracted and converted to electrons, then the full-well saturation flagging can't be performed (though the A-to-D saturation can still be).

Still working on getting the HRC image.

mdlpstsci, 2020-12-03:
@yotam-stsci 
For Item 1: Yes, I would rather use a "more official" file from you rather than my hacked version.  It also gives me a chance to check the keyword settings. Thanks for the image which I will collect in a bit.

For Item 3: I will make the logic for the full-well saturation flagging as you note.  The flagging for the A-to-D saturation is already set in this manner.

For Item 5: OK.

And for the HRC image - great.

mdlpstsci, 2020-12-08:
@yotam-stsci @mmcdonald123 @tddesjardins 
Ooops.  i do have a few more questions with regard to binning.

(A) Do you intend to make binned saturation images to accommodate binned ACS data?  I include Tyler here particularly because I do not know how often binned data is acquired.  Also, I do not know if it is fully accommodated in calacs.  I would assume that binned data IS processed in the same manner, after accounting for the binning, as the unbinned counterpart.  It looks like for BIAS correction, there do exist binned calibration images.  Only binned subarrays have to be handled in a special way in the pipeline for the BIAS correction.

(B) If you do not intend to generate binned calibration images, then I will need to explore further the binning routines used by the pipeline.

Thanks,
Michele

yotam-stsci, 2020-12-14:
Hi @mdlpstsci. Thanks for the further questions. I just got back from vacation, and will be working on this today and this week. Will get back to you ASAP.

yotam-stsci, 2020-12-14:
@mdlpstsci

The [ACS Data Handbook](https://hst-docs.stsci.edu/acsdhb/chapter-3-acs-calibration-pipeline/3-4-calacs-processing-steps) says that 

> ACS CCD data are not binned, so any image which is not the full size is assumed to be a subarray.

mdlpstsci, 2020-12-14:
@yotam-stsci 
If this is the case (and it seems to be), then this is great.

yotam-stsci, 2020-12-16:
@mdlpstsci 

I've put the full-frame WFC saturation map here:
`/user/ycohen/for_others/michele_dlp/acs_wfc_saturation_reffile_postsm4_fullframe.fits`

Going to try to get the HRC image today as well.

mdlpstsci, 2020-12-17:
@yotam-stsci @mmcdonald123 
Althought I cannot work on ACS projects until the beginning of
January, I took a look at the latest version of your saturation
image reference file so I could give you feedback ASAP.
I found the following issues:

1) Please Use the following values for The LTV1 and LTV2 in the SCI
   extensions.  This is because of the way the software is using
   this new file.
   LTV1 needs to be 24.0 for both extensions.
   LTV2 needs to be 0.0 for [SCI, 1].
   LTV2 needs to be 20.0 for [SCI, 2].

   Please note these values are float datatype.
   The LTV1 and LTV2 are the physical and virtual overscan sizes
   for full-frame images.

2) The DQ extensions should be integer (16 bits/pixel).

3) Do the SCI and ERR extensions really need to be
   64 bits/pixel (-64)?  If not, the extensions should only
   be 32 bits/pixel (-32).

4) Since this saturation image seems to be applicable only for
   post-SM4 data, you will need to make a comparable image for the
   pre-SM4 data.  Since the "old" software just used a constant
   for the saturation limit, I would suggest this would be a good
   value to use as a starting place. You can find the original
   static value(s) for the full well saturation in the CCDTAB.
   The ACS team might want to use a better/different value based
   upon your knowledge of the data.

5) Finally, I assumed your team discussed this, but there is
   a document (out-of-date for ACS) which discusses the form
   of the calibration/reference files.  This document can be found
   here: http://newcdbs.stsci.edu/doc/ICD47/Section1.html.  It needs
   to be updated (at the least) with the new SATUFILE information.
   This document hopefully also provides one with a more
   generalized view, for consistency, of all the calibration/reference
   files.

yotam-stsci, 2021-01-04:
@mdlpstsci 
Thanks for the notes and feedback! Just getting back from vacation today, so I'll be working on it this week. More soon...

yotam-stsci, 2021-01-05:
@mdlpstsci 

I have generated an updated file which takes care of your items 1-3. Regarding item 4: after discussing it with Norman, and doing some testing, we have decided that it should be fine to use the post-sm4 image for pre-sm4 data as well, at least for now. So I've removed the "postsm4" from the file name. You can find the new file here: `/user/ycohen/for_others/michele_dlp/acs_wfc_saturation_reffile_fullframe.fits`

Please let me know if there are any remaining modifications that need to be done.

Also, regarding item 5: We will be getting in contact with the relevant people about this. I will keep you posted.

mdlpstsci, 2021-01-06:
@yotam-stsci @mmcdonald123 

As far as proper working of the updated pipeline with the new saturation images, the only two items which you need to do are:
A. If the WFC saturation image is for pre- and post-SM4, you need to fix the USEAFTER date.
B. I still need an HRC image for testing.

Please know that I do not certify your files.  The changes I have requested are the modifications needed so the files work properly with the calibration pipeline code.  Things look quite good at this time.  Once I have an updated WFC file, and an HRC file that I can test, I will provide you with the updated software (or basically a viable executable on the Linux systems) which you can use for testing.

We are almost there!

yotam-stsci, 2021-01-11:
@mdlpstsci 
Thanks for the info.

I've updated the USEAFTER date for the WFC image, and put the new copy in the same place as before: `/user/ycohen/for_others/michele_dlp/acs_wfc_saturation_reffile_fullframe.fits`

I've also made a first version of the HRC file, which you can find in the same directory:
`/user/ycohen/for_others/michele_dlp/acs_hrc_saturation_reffile_fullframe.fits`

I'm not entirely certain that I got all the header keywords correct for the HRC file (I mostly copied them from a bias file), so I will make any necessary changes after you review it.

mdlpstsci, 2021-01-12:
@yotam-stsci @mmcdonald123 
The WFC looks to me.  The HRC file needs two fixes:

1. DQ extension should be 16 bits/pixel, and
2. EXTVER keyword and value is missing from all extensions.

yotam-stsci, 2021-01-12:
@mdlpstsci 
I think the HRC file should be fixed now. I've updated it at the same path as before.

mdlpstsci, 2021-01-12:
@yotam-stsci @mmcdonald123 
The HRC file looks good.  I will be emailing you instructions on how to access the calacs.e executable for testing.

mdlpstsci, 2021-02-02:
@yotam-stsci @mmcdonald123 
Can one of you add a comment here regarding your testing and acceptance of this code?  I am about to run the final developer regression tests and merge the code to prepare for the build.  Thanks.

yotam-stsci, 2021-02-02:
@mdlpstsci 
I wrote you an email last week regarding my testing and findings that the code was slower than I expected. Would you be able to comment on that before we proceed?

mdlpstsci, 2021-02-02:
@yotam-stsci @mmcdonald123 
The C pipelines are a very old code base written in a time when memory was costly.  As such, many routines work on a line-by-line basis.  I certainly agree the pipeline could be made more efficient in this modern era, but an upgrade should be a defined project of its own. At this time, the code uses library utilities specifically created for its support and is written in a consistent fashion for maintainability.

closed 2021-02-08 by mdlpstsci

mdlpstsci, 2021-02-10:
Slack conversation

On 08 February 2021 @yotam-stsci 
it looks good to me now!
execution time for that step is now just a few seconds, as expected, thanks so much for the fix

@mdlpstsci 
OK.  I will be starting the build process by 2:00 pm.  If you have any further objections, please say so, and I will hold off the build process.  Actually, you can always say so, there will just be a bit more cleanup.

This was the final conversation indicating it was fine to proceed with the build.

PR #539 (2021-02-08): Issue453 acs satmap
The CALACS pipeline has been updated to incorporate a full two-dimensional image to be used for full-well saturation flagging.  Essentially, the code which performed the full-well flagging using a single value in dodqi.c has been removed.  A new "step" has been added to the pipeline after BIAS correction, BLEV correction, and conversion to electrons - doFullWellSat().

This PR fixes #453.