spacetelescope/hstcal #460: A-to-D Saturation Flagging in ACS CCD Data

tddesjardins, 2019-12-05:
The ACS team would like to add to `calacs` a line that checks for pixels with >= 2^16 (65,536) DN during DQICORR and flags these with DQ flags 256 + 2048 (=2304). The ACS data handbook currently describes that this is happening, but it is not (it also says it happens during ATODOCORR, but that step has always been set to OMIT). 

If possible @mdlpstsci could we work this in to coincide with the darktime fix?

tddesjardins, 2019-12-06:
Actually, speaking with Norman, the limit should be 2^16 - 1 (65,635) DN.

tddesjardins, 2019-12-12:
Update: we've found a new issue where post-SM4 observations don't seem to follow the 65,535 limit rule. The value appears to be a little lower and it is not a single value, though the range in values is small. We would request that all pre-SM4 (before 2009-01-01) CCD observations use the 65,535 value, while AFTER SM4 pixels with 64,000 are flagged.

mdlpstsci, 2020-01-15:
@tddesjardins 

1. Can you please provide a dataset to use for testing?
2. I have done a quick check and the code is in place for the ACS CCDs to be flagged with (256 + 2048).  At this time, if the science pixel value is > 65534, then the current dq value is set to the current_value + 2048.  If the science pixel value is > 65534 or the CCD full well saturation level, then the current dq value is set to current_value + 256. The CCD full well saturation level is defined in the CCDTAB.

mdlpstsci, 2020-01-15:
Discussed with the ACS team that "calibration" values need to be incorporated into the calibration reference files. This is so the information is documented and readily accessible to the Instrument teams for future updates. The CCDTAB currently has a "full well saturation" column, SATURATE.  It is suggested another column be added to the CCDTAB which contains the A-to-D saturation value. Also, there should be a version of this file for pre- and post-SM4 data.  In this way the calibration of the data is reference-file driven and not hard-coded.

Just as FYI - the current code uses an A-to-D threshold of 65534, and the logic is such that values will be flagged with 2048 if the science pixel > 65534.

mdlpstsci, 2020-04-08:
Sent an email to interested parties on the ACS team for information on this ticket.  Response was:
From: Tyler Desjardins <desjard@stsci.edu>

> Date: Friday, March 27, 2020 at 1:50 PM
> To: Michele De La Pena <mdelapena@stsci.edu>
> Cc: Meaghan McDonald <mmcdonald@stsci.edu>, Yotam Cohen <ycohen@stsci.edu>, Norman A Grogin <nagrogin@stsci.edu>
> Subject: Re: GIt Issue 453 and 460
>  
> Hi Michele,

> CC’ing Meaghan and Norman to keep them in the loop, and Yotam since he’s involved in #453 (saturation maps).
>  
> For 460, I’ll update the CCDTABs again to include the A-to-D saturation limits for both WFC and HRC so there’s no difference in treatment between the CCDs. I can have that done early next week sometime as you’re right it’s a pretty simple fix. Meaghan will need to talk to Ed Slavich about updating the certification process for CCDTAB again and we’ll deliver them to the CRDS HST TEST server when they’re made and Ed’s ready.

mdlpstsci, 2020-04-08:
@tddesjardins @mmcdonald123 This sprint is over Friday, but I will carry forward this ticket to the new sprint starting on Monday if you think you will have new CCDTAB files that I can work on starting next week or the week after.

mdlpstsci, 2020-04-17:
On 17 April Meaghan wrote:
I’ve successfully delivered the updated CCDTAB files that contain the new ATODSAT column to HST TEST. They new files are:
44h17140j_ccd.fits
44h17142j_ccd.fits
44h17143j_ccd.fits
44h17145j_ccd.fits
44h17146j_ccd.fits
44h17148j_ccd.fits
44h17149j_ccd.fits
44h1714aj_ccd.fits
44h1714cj_ccd.fits
44h1714dj_ccd.fits
44h1714fj_ccd.fits
44h1714gj_ccd.fits
 
You can find them here: /grp/crds/hst/test/references/hst

closed 2020-05-15 by mdlpstsci

mdlpstsci, 2020-05-15:
@mmcdonald123 @tddesjardins Note this Issue/PR has completed and closed based on your comments that I added to the PR.  The software has been merged onto master and is slated to be in hstdp-2020.3 (date unknown at this time).