spacetelescope/hstcal #463: calwf3 does not produce any error if flat field file is not found

npirzkal, 2020-02-13:
When running calwf3 on a WFC3 IR dataset, if the flat field file listed under PFLTFILE is not found, calwf3 runs but does nothing, does nor report and error, does not explain why nothing was done and gives no explanation to the user as to why nothing was processed.

pllim, 2020-02-13:
What does the header look like? Hard to tell which of this logic below is hit without having access to the file in question:

https://github.com/spacetelescope/hstcal/blob/f60a04f5be97baf80b4fbfdd59fb520c81595284/pkg/wfc3/calwf3/wf3ir/getirflags.c#L404-L424

npirzkal, 2020-02-13:
Exactly like the original header except that the flat field file name is the name of a non existent file, e.g. iref$uc72113oi_pfl_patched2.fits

pllim, 2020-02-13:
Does it work if you rename it to something like `myflat_pfl.fits` and make sure it is indeed in where you define `iref`?

npirzkal, 2020-02-13:
No, it only works if the file is actually there. You should be able to test this easily enough.

pllim, 2020-02-13:
Maybe there is a bug in https://github.com/spacetelescope/hstcal/blob/master/pkg/wfc3/calwf3/calwf3/refexist.c 🤷‍♀ 

I'll let the WFC3 devs sort it out.

npirzkal, 2020-02-13:
Form the comments in that code, it seems so:

"
I made RefExist exit clean 
the first time it found a missing file, HSTIO was barfing any other way
"

pllim, 2020-02-13:
That was added by @sosey . Maybe she can comment.

mdlpstsci, 2020-02-16:
Sorry to come late to this party, but I have been out sick for most of last
week, and I am now just beginning to feel better.

First, let me ask you what version of calwf3 are you running, and are you using
the C executable or the Python interface?  If you are using the Python interface,
what version of the wfc3tools are you running?

Here are the steps I performed to try and diagnose this issue.  Please know that
I am using the versions of calwf3 (HSTCAL) and wfc3tools which are packaged as
hstdp-2019.5.  This is the same software which is used by the pipeline. You can see
the version numbers and dates of the C executable [Version 3.5.0 (Oct-09-2018)]
and wfc3tools [2.3.1 - hash: cb3ca69].

I did not have any issues properly processing the data.  However, if you like,
once I get your code and Python version numbers, I will try to reproduce your
problem.
Best,
Michele

--------------------------------------------------------
1) I chose a WFC3 IR image to process.
There are no FITS keywords corresponding to PFLTCORR, DFLTCORR, or LFLTCORR
in my raw file.  I can see these corrections are set to PERFORM in the
calwf3 code.  I can also see there is a valid reference file for PFLTFILE, and
both DFLTFILE and LFLTFILE are set "N/A". Under normal processing conditions using
the C executable, I get:
...
    FLATCORR PERFORM
    PFLTFILE iref$uc72113ni_pfl.fits
    PFLTFILE PEDIGREE=INFLIGHT 01/08/2009 30/11/2010
    PFLTFILE DESCRIP =Flat-field created from inflight observations (Aug 2009- Nov 2010).
    FLATCORR COMPLETE
...
*** WF3IR complete ***

=====

2) If I change the PFLTFILE value to a non-existent file, I get:

CALBEG*** CALWF3 -- Version 3.5.0(Oct-09-2018) ***
Begin    14-Feb-2020 16:47:26 EST

Input    idw101b5q_raw.fits
loading asn

LoadAsn:  Processing SINGLE exposure
Trying to open idw101b5q_raw.fits...
Read in Primary header from idw101b5q_raw.fits...
ERROR:     in HST I/O functions:
Filename iref$uc72113nx_pfl.fits EXTNAME  EXTVER 0 CFITSIO status 104
failed to find or open the following file: (ffopen)/grp/hst/cdbs/iref/uc72113nx_pfl.fits[0]
Error opening image array.

Warning    PFLTFILE, DFLTFILE, LFLTFILE are all blank
ERROR:    One reference file was missing.
ERROR:    Couldn't process IR data
ERROR:    CALWF3 processing NOT completed for idw101b5q_raw.fits
ERROR:    CALWF3 processing NOT completed for idw101b5q_raw.fits
ERROR:    Calibration file(s) missing or invalid.
ERROR:    Can't open file

==> Everything seems to be working as advertised from the C code.

=====

3) I invoke a Python 3.6 session...and run calwf3 on the same file as above with
the invalid pfltfile.
Python 3.6.8 |Anaconda, Inc.| (default, Dec 29 2018, 19:04:46)
[GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from wfc3tools import calwf3
The following tasks in the wfc3tools package can be run with TEAL:
  calwf3    pstack    pstat     wf32d     wf3ccd    wf3cte    wf3ir     wf3rej
>>> calwf3("idw101b5q_raw.fits")
git tag: 2.3.1

git branch: HEAD

HEAD @: cb3ca69be9eb334780b68178fcf737e3e0b44de8

CALBEG*** CALWF3 -- Version 3.5.0(Oct-09-2018) ***

Begin    16-Feb-2020 13:14:15 EST

Input    idw101b5q_raw.fits

loading asn

LoadAsn:  Processing SINGLE exposure

Trying to open idw101b5q_raw.fits...

Read in Primary header from idw101b5q_raw.fits...

ERROR:     in HST I/O functions:

Filename iref$uc72113nx_pfl.fits EXTNAME  EXTVER 0 CFITSIO status 104

failed to find or open the following file: (ffopen)/grp/hst/cdbs/iref/uc72113nx_pfl.fits[0]

Error opening image array.

Warning    PFLTFILE, DFLTFILE, LFLTFILE are all blank

ERROR:    One reference file was missing.

ERROR:    Couldn't process IR data

ERROR:    CALWF3 processing NOT completed for idw101b5q_raw.fits

ERROR:    CALWF3 processing NOT completed for idw101b5q_raw.fits

ERROR:    Calibration file(s) missing or invalid.

ERROR:    Can't open file

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/mdelapena/miniconda3/envs/hstdp-2019.5/lib/python3.6/site-packages/wfc3tools/calwf3.py", line 79, in calwf3
    raise RuntimeError("calwf3.e exited with code {}".format(ec))
RuntimeError: calwf3.e exited with code ERROR_RETURN

Here the Python interface as faithfully generated the low-level C errors.

mdlpstsci, 2022-01-11:
@pllim Do you think I can close this now?  I see the appropriate errors as noted above, and Nor never responded.

pllim, 2022-01-11:
@mdlpstsci , I think that is your call. If this is treated as a help call, it is fine to close as "won't fix" when there is no reply from the user after some time. I think two years is enough time?

mdlpstsci, 2022-01-11:
As shown above, the appropriate errors are generated in at least CALWF3 V3.5.0 and newer so this should no longer be an issue.   Also, the user never responded.

Will not do.

closed 2022-01-11 by mdlpstsci