spacetelescope/hstcal #474: WFC3 crj file update

mdlpstsci, 2020-04-07:
WFC3 would like to fix the bug that the IR crj files are in electrons and need to be converted to electrons per second. 
We recommend to change the code to put the data in units of e/s  but this is a low priority and we will not be making any other changes to the crj code at this time.  

Cross reference to JIRA HLA-310

mackjenn, 2025-10-27:
Hi.  Adding a quick note that IR crj files are repeat-obs (not CR-split) and they are very common.  

See this section of the DHB. 
https://hst-docs.stsci.edu/wfc3dhb/chapter-3-wfc3-data-calibration/3-3-ir-data-calibration-steps#id-3.3IRDataCalibrationSteps-3.3.133.3.13Cosmic-rayrejection

These files should be in the same units as flt files, which is electrons per second.  If they are not, this would need to be fixed.

Jen

pllim, 2025-10-27:
Oh, IR! OK I am not familiar with that one. Thanks for the clarification.