spacetelescope/hstcal #489: Post-flashed ACS CRJ/CRC images are not fully flash-corrected

jryon, 2020-05-18:
Post-flashed CRJ/CRC images show residual flash background after flash correction. The FLASHDUR keyword is the same in the FLT of a single CR-SPLIT as in the CRJ of the sum of CR-SPLITs.

The solution should be to sum the FLASHDUR keywords of the input images to acsrej, and save the sum to the FLASHDUR keyword of the output CRJ/CRC. The FLASHDUR keyword is used directly in acs2d to scale the flash reference file for flash correction.

mdlpstsci, 2021-04-30:
@jryon Do you have a dataset that would be good for testing this change?

jryon, 2021-04-30:
@mdlpstsci  jdzn01b6q and jdzn01b7q could be used to test. They are post-flashed, CR-split images, and also 1k subarrays. I can look for some full-frame data if you'd prefer that. Their association ID is jdzn01010, if you need it.

mdlpstsci, 2021-04-30:
@jryon Thanks.  Let me start with this association.

mdlpstsci, 2021-05-03:
@jryon Before I make the software available for you for any testing you might like to do, I have put a tar file on plhstins1 in /home/mdelapena/FOR_JENNA/imp_flash_test.tar.  This file contains the "jdz" data processed with the current release of CALACS (BADFLASH subdirectory) and the modified CALACS (NEWFLASH subdirectory).  Please see if this newly processed data meets your expectations.

jryon, 2021-05-03:
@mdlpstsci Thanks! The NEWFLASH data meets expectations. The FLASHDUR keyword in the NEWFLASH CRJ is the sum of the individual exposure FLASHDURs, and there is no obvious post-flash signal remaining in the NEWFLASH CRJ (such signal is clearly visible in the BADFLASH CRJ). I also measured a sigma-clipped median of the background with the bright star masked out for both CRJs: BADFLASH median = 33.8 e-, NEWFLASH median = 1.7 e-. The remaining signal in the NEWFLASH CRJ is likely to be sky background, but I can investigate that more closely when the software is available.

mdlpstsci, 2021-05-03:
@jryon I have put the "test" executable for calacs.e which incorporates the flash duration fix on plhstins1.  You should just be able to invoke my executable (!) and process data.  In the past I have had folks do a bit more setup, but I am not sure this was really necessary.  My executable is where CALACS is Version 10.3.3 (03-May-2021):
/home/mdelapena/repos_hstcal/hstcal/bin.Linux-2.6.32-754.36.1.el6.x86_64-x86_64-with-redhat-6.10-Santiago/calacs.e

jryon, 2021-05-06:
@mdlpstsci Thanks! I ran the executable you shared on a larger set of post-flashed CR-SPLIT data. I created a 2D model for the background in each CRJ and plotted the median of the model as a function of exposure time for each image. The backgrounds are all <5e-, which makes sense for their short exposure times. There's a bit of trend towards higher backgrounds with longer exposure times, which also makes sense if the remaining signal is sky background. I looked at each CRJ's 2D model background, and I didn't see any remaining post-flash gradient. Given all of this, I'm happy to sign off on this fix.

mdlpstsci, 2021-05-06:
@jryon  Cool. I will generate a PR from this code change so the code can be reviewed and you can add your commentary.

mdlpstsci, 2021-05-06:
@jryon Actually, please add your commentary approving the changes directly to this issue.  Thanks!

jryon, 2021-05-06:
I approve these changes. 

Please let me know if I need to give more detail.

mdlpstsci, 2021-05-12:
Resolved by PR#550.

closed 2021-05-12 by mdlpstsci