spacetelescope/hstcal #490: ACS CRJ/CRC images are not fully dark-corrected

jryon, 2020-05-18:
In the verbose output from acs2d, the MEANDARK value between an FLT CR-SPLIT and its associated CRJ are the same, though the CRJ is the sum of the CR-SPLITs. This means that dark correction is using the same darktime to scale the dark reference frame for both. The darktime used by acs2d is the sum of the header keyword DARKTIME and a commanding overhead read from CCDTAB. The keyword DARKTIME in a CRJ is not the sum of the input CR-SPLITs' DARKTIMEs, it is the same as the first input CR-SPLIT. Similarly, in acs2d, the commanding overhead is not scaled by the number of images summed to create the input CRJ.

Rather than apply the commanding overhead in acs2d, add it to the DARKTIME keyword during acsccd, so every BLV_TMP file has the correct DARKTIME keyword. In acsrej, sum the input CR-SPLITs' DARKTIME keywords and set the CRJ/CRC DARKTIME keyword to this sum. In acs2d, use ONLY the DARKTIME keyword of the input file to scale the dark reference file for dark correction.

mdlpstsci, 2021-05-25:
@jryon I have implemented a fix to address this issue, but I now need you to verify that I actually did what you wanted!  I have put "old, bad test files", as well as the "new, good test files" on plhstins2.  The files are in /home/mdelapena/FOR_JENNA in directories BADDARK and NEWDARK.  You can also run the executable on data of your choosing for testing.  The executable is located here:
/home/mdelapena/repos_hstcal/hstcal/bin.Linux-2.6.32-754.36.1.el6.x86_64-x86_64-with-redhat-6.10-Santiago/calacs.e

The version is "10.3.3 (24-May-2021)".  This is the same numeric version as the fix for the flash correction, but the date has been updated.

jryon, 2021-05-26:
@mdlpstsci I tested the data in BADDARK and NEWDARK, and used the executable you provided to test a set of ~1300s full-frame CR-SPLIT data from program 15940. For the data processed with the fix, the commanding times from the CCDTAB are added to the DARKTIME keywords by acsccd, and the DARKTIME keywords of the input files to acsrej are summed together, both as requested. The new trailer files contain MEANDARK values that are ~2x larger than in the bad trailer files, which is expected because the DARKTIME keywords ~doubled in all cases. I measured median backgrounds in the new and bad data, and found that the differences agree with expectations given the increased DARKTIME of the new data. Visually, the data from program 15940 shows a marked improvement in dark correction. In this screenshot, the left image is new data, the right image is bad data, and both scales are the same:

[image]

One question, are the MEANDARK values calculated as a straight mean of _all_ pixels in the SCI array of superdark*DARKTIME? Or are hot/warm pixels and other flagged pixels ignored? The MEANDARK values I'm seeing in all of the trailer files (bad and new) is lower than I'd expect given the mean dark rates (including all pixels) of our superdarks.

mdlpstsci, 2021-05-26:
@jryon The MEANDARK value reported in the trailer files is the average of all the good pixels in the image data. By this I mean if the DQ value for a science pixel is set to **anything** (2ˆ16 - 1 or 65535) then the SCI pixel is not included in the sum of good pixel values.  

The computation of MEANDARK is different than it used to be.  Formerly the computations were done line by line.  A mean and a weight (num_good_pix/total_pix_in_line) were computed. A running weighted mean was then computed, as well as a running weight.  The last step was to compute the mean of the running value.  The pseudo-code below probably describes the situation better:
 
for i = 1, all_lines
    mean += mean_of_line * weight_of_line;
    weight += weight_of_line;

final_mean = mean/weight

If you would like the computation of the MEANDARK value to be changed, just let me know.

jryon, 2021-05-26:
@mdlpstsci Thanks for the explanation! The MEANDARK values make perfect sense now. No need to change the MEANDARK calculation. I just wanted to make sure I understood their origin so that I could confirm the amount of dark current removed in DARKCORR is expected given the CRJ DARKTIME keywords.

I approve this bugfix.

mdlpstsci, 2021-05-26:
@jryon Cool.  I need to know if you feel comfortable having this issue, as well as Issue#489, in the **_same_** build as they both affect the CRJ/CRC files?  If you feel that you can distinguish between the updates to validate the build, then both of these fixes can be submitted together.  If not, then Issue#489 is already on master and ready to go, so that fix will be in the next build.

jryon, 2021-05-26:
@mdlpstsci I'm comfortable having both issues in the same build, thanks!

mdlpstsci, 2021-05-26:
@jryon SUPER.  I will get the code ready for a build so as soon as the opportunity becomes available, the fixes will be submitted.

closed 2021-06-04 by mdlpstsci