spacetelescope/hstcal #509: TST: stis.test_lev1_fuvspec chronically failing

pllim, 2020-07-27:
After getting spammed by the bot for a few weeks, I decided to look at the log even though I am not part of this project anymore. Looks like a STIS test is failing. Someone should look into it. Example log at #508. 

```
self = <tests.stis.test_lev1_fuvspec.TestLev1FUVSpec object at 0x7f1951cf7898>

    def test_lev1(self):
        """This was ``calstis_lev1_FUVspec``."""
        raw_file = 'o5cl02040_raw.fits'
        wav_file = 'o5cl02040_wav.fits'
    
        # Prepare input files.
        self.get_input_file(raw_file)
        get_bigdata('scsb-hstcal', self.env, 'stis', 'fuv-mama', 'input',
                    wav_file)
    
        # Run CALSTIS (equivalent to stistools.calstis.calstis)
        subprocess.call(['cs0.e', raw_file, '-v'])
    
        # Compare results
        outputs = [('o5cl02040_flt.fits', 'calstis_lev1_FUVspec_flt.fits'),
                   ('o5cl02040_x1d.fits', 'calstis_lev1_FUVspec_x1d.fits')]
>       self.compare_outputs(outputs)

/srv/jenkins/workspace/RT/hstcal/clone/tests/stis/test_lev1_fuvspec.py:27: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <tests.stis.test_lev1_fuvspec.TestLev1FUVSpec object at 0x7f1951cf7898>
outputs = [('o5cl02040_flt.fits', 'calstis_lev1_FUVspec_flt.fits'), ('o5cl02040_x1d.fits', 'calstis_lev1_FUVspec_x1d.fits')]
atol = 0, rtol = 1e-07, raise_error = True, ignore_keywords_overwrite = None
verbose = True

    def compare_outputs(self, outputs, atol=0, rtol=1e-7, raise_error=True,
                        ignore_keywords_overwrite=None, verbose=True):
        """
        Compare CALXXX output with "truth" using ``fitsdiff``.
    
        Parameters
        ----------
        outputs : list of tuple
            A list of tuples, each containing filename (without path)
            of CALXXX output and truth, in that order. Example::
    
                [('output1.fits', 'truth1.fits'),
                 ('output2.fits', 'truth2.fits'),
                 ...]
    
        atol, rtol : float
            Absolute and relative tolerance for data comparison.
    
        raise_error : bool
            Raise ``AssertionError`` if difference is found.
    
        ignore_keywords_overwrite : list of str or `None`
            If not `None`, these will overwrite
            ``self.ignore_keywords`` for the calling test.
    
        verbose : bool
            Print extra info to screen.
    
        Returns
        -------
        report : str
            Report from ``fitsdiff``.
            This is part of error message if ``raise_error=True``.
    
        """
        all_okay = True
        creature_report = ''
        updated_outputs = []  # To track outputs for Artifactory JSON schema
    
[... 166 lines cut ...]
E                         a> 3.7876737e-07
E                          ?       ^^^
E                         b> 3.7876686e-07
E                          ?       ^^^
E                       ...and at 970 more indices.
E                    Column OFFSET data differs in row 10:
E                       a> 4.1717997
E                        ?         -
E                       b> 4.171799
E                    Column OFFSET data differs in row 22:
E                       a> -1.0231307
E                        ?          ^
E                       b> -1.0231305
E                        ?          ^
E                    Column OFFSET data differs in row 29:
E                       a> -0.109361395
E                        ?          ^^
E                       b> -0.109361485
E                        ?          ^^
E                    ...187 additional difference(s) found.
E                    ...
E                    197 different table data element(s) found (31.42% different).

/srv/jenkins/workspace/RT/hstcal/clone/tests/helpers.py:300: AssertionError
```

mdlpstsci, 2020-07-27:
The last commit was on 07 July for some clean-up associated with ACS (only) copyrighted code changes, and there were no commits after this date.  As far as I can tell, the first indication/notification of a problem with the STIS test was on 11 July.  I do not know how to look further back in the logs to see the last clean run.  Having said this, you indicated you ran the "old" and "new" code on RAW data and got compatible results (i.e., no issues).  Just let me confirm that you ran the same regression test as the one failing and using the same dataset?  Do you store any of the data used for the regression tests on Artifactory?  If so, can you download the files and make sure they match the data you produced when trying to diagnose this issue?

pllim, 2020-07-27:
I don't have any WBS assigned to work on this. I am hoping @stscirij can look into it.

stscirij, 2020-07-27:
Run 697 on July 11 was OK, run 698 on July 12 failed.  The last commit to the hstcal repo was on July 7, the commit that had the numerical recipes changes was on June 25.  The truth files are 2 years old.

The test just runs cs0.e on a single science/wavecal dataset pair and compares the output flt and x1d files.  First I retrieved the data from the archive and ran cs0.e using an old version (dated 19 Jan 2018).  Then I pulled from master and build the code and ran again, and got identical results.  I also pulled the input and truth files from artifactory and ran the old and new versions of the code and got the same results using the old code and the new code, which is different from the truth files.  So I don't understand what happened...

Robert J.

stscirij, 2020-07-27:
Am I correct in saying that this code is now released and running in the pipeline?  Did the STIS team sign off on the test products?  IF so then perhaps we should just update our truth files and be done with it.

Robert J>

________________________________________
From: Robert Jedrzejewski
Sent: Monday, July 27, 2020 1:55:38 PM
To: spacetelescope/hstcal; spacetelescope/hstcal
Cc: Assign
Subject: Re: [spacetelescope/hstcal] TST: stis.test_lev1_fuvspec chronically failing (#509)

Run 697 on July 11 was OK, run 698 on July 12 failed.  The last commit to the hstcal repo was on July 7, the commit that had the numerical recipes changes was on June 25.  The truth files are 2 years old.

The test just runs cs0.e on a single science/wavecal dataset pair and compares the output flt and x1d files.  First I retrieved the data from the archive and ran cs0.e using an old version (dated 19 Jan 2018).  Then I pulled from master and build the code and ran again, and got identical results.  I also pulled the input and truth files from artifactory and ran the old and new versions of the code and got the same results using the old code and the new code, which is different from the truth files.  So I don't understand what happened...

Robert J.

mdlpstsci, 2020-07-27:
HI,
Yes this code was released (final release: HCALDP_20200708_cal).  Matt Maclay performed a number of tests to validate STIS.

Michele

Michele D. De La Pena
Space Telescope Science Institute
mdelapena@stsci.edu<mailto:mdelapena@stsci.edu>
Mobile: 520.419.8825
Office: 667.218.6501

mdlpstsci, 2020-07-29:
I talked to Warren today about this issue.  Considering: 

- you did tests with "old" and "new" versions of the cs0.e on known input files and generated matching output files,
- tests were done by M. Maclay and he signed off on the build, 
- you probably no longer know the pedigree of the truth files on Artifactory, and
- the failure seems to have happened "out of the blue" associated with no known changes to the repo, 

then perhaps, indeed, it would be best to generate new truth files.

pllim, 2020-08-10:
Please close if this has been fixed. Thanks!

stscirij, 2020-08-10:
New truth files uploaded to artifactory

closed 2020-08-10 by stscirij