spacetelescope/hstcal #517: Windows support

pllim, 2020-08-10:
There was a request for Windows support. User wants to use `stistools` on Windows, and it calls CALSTIS executables from here.

ref: INC0157637

Also see spacetelescope/stistools#128

zacharyburnett, 2023-04-20:
a user also asked for Windows support for `stenv`, which is prevented from supporting Windows by `hstcal` and `fitsverify`: https://github.com/spacetelescope/stenv/issues/100

Would it be much work to modify the `conda-forge` recipe to include a Windows build? I imagine it might require AppVeyor or something similar to have access to the MSVC in order to build. Supporting Windows might require a lot of maintenance.

pllim, 2023-04-20:
I don't think it is happening. Better to ask stenv not to install those packages on Windows.

pllim, 2023-04-20:
So I think we're discussing different motivations on this same issue:

* My original request was from someone specifically wanting to run calstis on Windows. We did nothing.
* Now someone wants to "install" stenv on Windows and it happens to grab hstcal. The user has not mentioned they want to actually run hstcal. So there is zero motivation here to get hstcal on Windows now. Much easier to not have stenv grab hstcal when Windows is detected.

zacharyburnett, 2023-04-20:
> * Now someone wants to "install" stenv on Windows and it happens to grab hstcal. The user has not mentioned they want to actually run hstcal. So there is zero motivation here to get hstcal on Windows now. Much easier to not have stenv grab hstcal when Windows is detected.

That's a good point; I can inform them that they can modify the environment to remove `hstcal` and `fitsverify`; however, it is also not guaranteed that the JWST pipeline will work on Windows either.

pllim, 2024-02-22:
Yeah, it is not happening. Even if we get this package compatible, CRDS is a different story.

closed 2024-02-22 by pllim