spacetelescope/hstcal #544: ACSCTE segfaulting in HSTCAL 2.6.0

mmcdonald123, 2021-04-06:
After updating to HSTCAL 2.6.0, which includes CALACS 10.3.0, the ACSCTE function in CALACS is not functioning properly. The issue was first noticed when trying to run the ACS Reference File Pipeline. After updating the pipeline environment to include CALACS 10.3.0, the pipeline repeatedly failed on the first instance where ACSCTE is called. The pipeline failed for reference files for the 2021-03-05 anneal (most recent) and also when testing the 2021-02-01 anneal (which successfully ran last month using CALACS 10.2.4). The type of error message I received for the failure was as follows:

`subprocess.CalledProcessError: Command '['acscte.e', '/grp/hst/acs7/reffile_pipe_devtest/REFERENCE_FILES/20210201/WFC/WFC/2.0/ABCD/20210201_dark_crrej.fits']' died with <Signals.SIGSEGV: 11>.`

To test the ACSCTE functionality outside the pipeline, @jryon and I ran `acscte.e` on the command line using CALACS 10.3.0 and also CALACS 10.2.4 for comparison. The results are as follows:

**Version 10.3.0**
`(test_hstcal) ryon@plhstins3:/grp/hst/acs3/ryon/test_hstcal$ acscte.e`
`Segmentation fault (core dumped)`
`(test_hstcal) ryon@plhstins3:/grp/hst/acs3/ryon/test_hstcal$ acscte.e je3w05vcq_blc_tmp.fits`
`Segmentation fault (core dumped)`
`(test_hstcal) ryon@plhstins3:/grp/hst/acs3/ryon/test_hstcal$ calacs.e -v`
`CALACS Version 10.3.0`
`syntax:  calacs.e [--help] [-t] [-s] [-v] [-q] [-r] [--version] [--gitinfo] [-1|--nthreads <N>] [--ctegen <1|2>] [--pctetab <path>] input`

**Version 10.2.4**
`(astroconda) ryon@plhstins3:/grp/hst/acs3/ryon/test_hstcal$ acscte.e`
`syntax:  acscte.e [--help] [-t] [-v] [-q] [--version] [--gitinfo] [-1|--nthreads <N>] [--ctegen <1|2>] [--pctetab <path>] [--forwardModelOnly] input [output]`
`(astroconda) ryon@plhstins3:/grp/hst/acs3/ryon/test_hstcal$ calacs.e -v`
`CALACS Version 10.2.4`
`syntax:  calacs.e [--help] [-t] [-s] [-v] [-q] [-r] [--version] [--gitinfo] [-1|--nthreads <N>] [--ctegen <1|2>] [--pctetab <path>] input `
`(astroconda) ryon@plhstins3:/grp/hst/acs3/ryon/test_hstcal$ acscte.e je3w05vcq_blc_tmp.fits`
`git tag: 2.5.0`
`git branch: HEAD`
`HEAD @: e08676b02e4c7cd06e3d5630b62f7b59951ac8c3`
`Setting max threads to 32 out of 32 available`
`Trying to open je3w05vcq_blc_tmp.fits...`

It seems that running `acscte.e` in the updated environment leads to a "Segmentation fault" error. I've reproduced this issue using several files that have been successfully CTE corrected with the previous version of CALACS including:

-  `/user/mmcdonald/anneal_analysis/test/ANNEAL_DARK/20210305/pre/pre_20210305_drk.fits`

- `/user/mmcdonald/anneal_analysis/test/ANNEAL_DARK/20210305/post/post_20210305_drk.fits`

- Any of these from the 2021-02-01 anneal period: `/user/mmcdonald/test/*blv_tmp.fits`

As it stands right now, this issue is preventing the ACS team from producing necessary dark, bias and sink pixel reference files.

mmcdonald123, 2021-04-06:
Hi @mdlpstsci , here is the ticket with details on the acscte issue.

mdlpstsci, 2021-04-07:
I have done a check on running acscte.e directly from the conda distribution (caldp_20210323) on both MacOS and Linux, and it generates a segfault.  I will note this was not caught by the regression testing as only the main executable for acs, calacs.e, is put through all its paces (many variations) as it is the executable run by the official pipelines.  In any case, in order to debug the issue, I pulled the code from the release via its tag and compiled it.  Unfortunately, when I compile the code on the local machines the executables which are generated all run just fine.

mdlpstsci, 2021-04-09:
The problem was discovered and addressed by PRs #545 and #546.  The code was tested on Linux and MacOS and was used to generate the Release Candidate 3 of the WFC3 CTE Build (HSTCAL Version 2.7.0 Release Candidate 3 [HSTDP-2021.2.0]).  The Release Candidate 3 has been delivered to operations for testing.

It should be noted in the interim, I recompiled the code to provide the ACS team with a working version so they could continue with their work.

mdlpstsci, 2021-04-12:
Resolved by PR #545 and #546.

closed 2021-04-12 by mdlpstsci