spacetelescope/hstcal #549: Convert CALACS components from working line-by-line to entire image

mdlpstsci, 2021-05-04:
This issue is intentionally a bit vague in that the specific components of the CALACS pipeline need to be analyzed for the opportunity to convert the component from line-by-line processing to working with the entire image.  If a component is identified as being appropriate for the conversion, a specific Issue should be created.  The goal of the conversion is to make CALACS easier to read, maintain, streamlined, and more efficient in terms of processing time.  Of course, any conversion needs to be balanced against memory constraints and priority of the task versus improvements which lead to higher quality science products.

pllim, 2021-05-04:
Hmmm... Might as well rewrite in Python? Numpy array manipulations use C vectorization anyway.

mdlpstsci, 2022-01-12:
I set this to low priority as the current version of CALACS works fine, and there is other work (new algorithm implementations and bug fixes) which should have higher priority.

mdlpstsci, 2023-09-05:
I am closing this issue as it is an issue of **_resources_** to work on something which is not actually a problem, per se.

closed 2023-09-05 by mdlpstsci