spacetelescope/hstcal #554: acsccd.e crashes when input is too long

pllim, 2021-06-28:
As discovered by @samanthalh when running it with `acstools.acsccd.acsccd('*raw.fits')` on too many files. That command expands to `acsccd.e file1,file2,file3,...` . Also see INC0169541 . `acsccd.e` crashes ungracefully with no helpful error message.

Offending lines:

https://github.com/spacetelescope/hstcal/blob/d395b8b88c2f4ccad65c6a3f176bf1907066c12d/pkg/acs/calacs/acsccd/mainccd.c#L106

https://github.com/spacetelescope/hstcal/blob/e0f20b7810862ed790e9c4989798eb51a8ec5bc1/include/hstcal.h#L11

mdlpstsci, 2021-07-16:
There are two options here: enlarge the length of the string acceptable as the "input" (and "output") or to trap the problem and issue a human understandable message.  I have chosen to do the latter as an ever larger string can always be found to be "not large enough".  Note this same problem permeates all of the main routines for the standalone components of the ACS pipeline.

samanthalh, 2021-07-19:
An error message that lets the problem be quickly diagnosed and fixed would be a big improvement! Thanks @mdlpstsci!!

pllim, 2021-07-19:
> trap the problem

👍 on this solution.

pllim, 2021-07-19:
p.s. Oof, my C is rusty... By trapping, do you mean force a `\0` at allowed max length and then check if the input ends with `.fits`?

pllim, 2021-07-21:
With help from @jhunkeler , I am able to obtain the stack trace as follows... This is more like a note for my future self, but hope it helps anyway!

1. Install HSTCAL with debug symbols.

```
./waf configure --prefix=... --debug
./waf build
./waf install
```

2. Have some data ready, like a lot of RAW files for ACSCCD.

3. In the working dir where the data are, fire up `gdb`.

4. Run offending command inside `gdb`.

```
(gdb) file .../bin/acsccd.e
Reading symbols from .../bin/acsccd.e...done.
```
```
(gdb) run jbbgcadmq_raw.fits,jbbgcbdsq_raw.fits,jbbgdai1q_raw.fits,jbbgdbieq_raw.fits,jbbgeajiq_raw.fits,jbbgebjoq_raw.fits,jbbgfac6q_raw.fits,jbbgfbciq_raw.fits,jbbggaohq_raw.fits,jbbggbotq_raw.fits,jbbghas7q_raw.fits,jbbghbsfq_raw.fits,jbbgiadjq_raw.fits,jbbgibdtq_raw.fits,jbbgjaf7q_raw.fits,jbbgjbfgq_raw.fits,jbbgkafiq_raw.fits,jbbgkbfmq_raw.fits,jbbglalxq_raw.fits,jbbglbm6q_raw.fits,jbbgmauiq_raw.fits,jbbgmbvzq_raw.fits,jbbgnaahq_raw.fits,jbbgnbarq_raw.fits
Starting program: .../bin/acsccd.e jbbgcadmq_raw.fits,jbbgcbdsq_raw.fits,jbbgdai1q_raw.fits,jbbgdbieq_raw.fits,jbbgeajiq_raw.fits,jbbgebjoq_raw.fits,jbbgfac6q_raw.fits,jbbgfbciq_raw.fits,jbbggaohq_raw.fits,jbbggbotq_raw.fits,jbbghas7q_raw.fits,jbbghbsfq_raw.fits,jbbgiadjq_raw.fits,jbbgibdtq_raw.fits,jbbgjaf7q_raw.fits,jbbgjbfgq_raw.fits,jbbgkafiq_raw.fits,jbbgkbfmq_raw.fits,jbbglalxq_raw.fits,jbbglbm6q_raw.fits,jbbgmauiq_raw.fits,jbbgmbvzq_raw.fits,jbbgnaahq_raw.fits,jbbgnbarq_raw.fits
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
git tag: 2.3.2rc1-41-gd395b8b
git branch: master
HEAD @: d395b8b88c2f4ccad65c6a3f176bf1907066c12d
You specified 24 input files but 10 output files.
ERROR:    Can't open file 
*** Error in `.../bin/acsccd.e': munmap_chunk(): invalid pointer: 0x0000000000637200 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7f474)[0x7ffff704b474]
.../bin/acsccd.e[0x41860b]
.../bin/acsccd.e[0x41875b]
.../bin/acsccd.e[0x418812]
.../bin/acsccd.e[0x402534]
/lib64/libc.so.6(__libc_start_main+0xf5)[0x7ffff6fee555]
.../bin/acsccd.e[0x401e79]
======= Memory map: ========
00400000-00435000 r-xp 00000000 fd:05 1174772647                         .../bin/acsccd.e
00634000-00635000 r--p 00034000 fd:05 1174772647                         .../bin/acsccd.e
00635000-00637000 rw-p 00035000 fd:05 1174772647                         .../bin/acsccd.e
00637000-00658000 rw-p 00000000 00:00 0                                  [heap]
7ffff675f000-7ffff6763000 rw-p 00000000 00:00 0 
7ffff6763000-7ffff6779000 r-xp 00000000 fd:02 8494712                    /usr/lib64/libresolv-2.17.so
7ffff6779000-7ffff6979000 ---p 00016000 fd:02 8494712                    /usr/lib64/libresolv-2.17.so
7ffff6979000-7ffff697a000 r--p 00016000 fd:02 8494712                    /usr/lib64/libresolv-2.17.so
7ffff697a000-7ffff697b000 rw-p 00017000 fd:02 8494712                    /usr/lib64/libresolv-2.17.so
7ffff697b000-7ffff697e000 rw-p 00000000 00:00 0 
7ffff697e000-7ffff6980000 r-xp 00000000 fd:02 8481105                    /usr/lib64/libdl-2.17.so
7ffff6980000-7ffff6b80000 ---p 00002000 fd:02 8481105                    /usr/lib64/libdl-2.17.so
7ffff6b80000-7ffff6b81000 r--p 00002000 fd:02 8481105                    /usr/lib64/libdl-2.17.so
7ffff6b81000-7ffff6b82000 rw-p 00003000 fd:02 8481105                    /usr/lib64/libdl-2.17.so
7ffff6b82000-7ffff6b88000 r--p 00000000 fd:05 1173392652                 .../lib/libk5crypto.so.3.1
7ffff6b88000-7ffff6b94000 r-xp 00006000 fd:05 1173392652                 .../lib/libk5crypto.so.3.1
7ffff6b94000-7ffff6b98000 r--p 00012000 fd:05 1173392652                 .../lib/libk5crypto.so.3.1
7ffff6b98000-7ffff6b9a000 r--p 00015000 fd:05 1173392652                 .../lib/libk5crypto.so.3.1
7ffff6b9a000-7ffff6b9b000 rw-p 00017000 fd:05 1173392652                 .../lib/libk5crypto.so.3.1
7ffff6b9b000-7ffff6b9c000 rw-p 00000000 00:00 0 
7ffff6b9c000-7ffff6bc0000 r--p 00000000 fd:05 1173501314                 .../lib/libkrb5.so.3.3
7ffff6bc0000-7ffff6c17000 r-xp 00024000 fd:05 1173501314                 .../lib/libkrb5.so.3.3
7ffff6c17000-7ffff6c5e000 r--p 0007b000 fd:05 1173501314                 .../lib/libkrb5.so.3.3
7ffff6c5e000-7ffff6c5f000 ---p 000c2000 fd:05 1173501314                 .../lib/libkrb5.so.3.3
7ffff6c5f000-7ffff6c6d000 r--p 000c2000 fd:05 1173501314                 .../lib/libkrb5.so.3.3
7ffff6c6d000-7ffff6c6f000 rw-p 000d0000 fd:05 1173501314                 .../lib/libkrb5.so.3.3
7ffff6c6f000-7ffff6cea000 r--p 00000000 fd:05 3257267045                 .../lib/libcrypto.so.1.1
7ffff6cea000-7ffff6e7d000 r-xp 0007b000 fd:05 3257267045                 .../lib/libcrypto.so.1.1
7ffff6e7d000-7ffff6f0a000 r--p 0020e000 fd:05 3257267045                 .../lib/libcrypto.so.1.1
7ffff6f0a000-7ffff6f35000 r--p 0029a000 fd:05 3257267045                 .../lib/libcrypto.so.1.1
[... 60 lines cut ...]
(gdb) p freeOnExit::reg[0]
$1 = {cursor = 8, length = 11, ptrs = 0x637030, freeFunctions = 0x637090}
(gdb) x /s freeOnExit::reg->ptrs[0]
0x0:	<Address 0x0 out of bounds>
(gdb) x /s freeOnExit::reg->ptrs[1]
0x6370f0:	"jbbgcadmq_raw.fits,jbbgcbdsq_raw.fits,jbbgdai1q_raw.fits,jbbgdbieq_raw.fits,jbbgeajiq_raw.fits,jbbgebjoq_raw.fits,jbbgfac6q_raw.fits,jbbgfbciq_raw.fits,jbbggaohq_raw.fits,jbbggbotq_raw.fits,jbbghas7q_"...
(gdb) x /s freeOnExit::reg->ptrs[2]
[... cut ...]
(gdb) x /s freeOnExit::reg->ptrs[3]
0x637310:	""
(gdb) x /s freeOnExit::reg->ptrs[4]
0x637420:	""
(gdb) x /s freeOnExit::reg->ptrs[5]
0x7fffffffd710:	"dummy"
(gdb) x /s freeOnExit::reg->ptrs[6]
0x6360c0 <trlbuf>:	""
(gdb) x /s freeOnExit::reg->ptrs[7]
0x637650:	"\200vc"
(gdb) x /s freeOnExit::reg->ptrs[8]
0x637a30:	"`zc"
(gdb) x /s freeOnExit::reg->ptrs[9]
0x18b58fb:	<Address 0x18b58fb out of bounds>
(gdb) x /s freeOnExit::reg->ptrs[10]
0x400746d73040018:	<Address 0x400746d73040018 out of bounds>
```

PR #556 (2021-07-30): Use strncpy for inlist in acsccd.e, better error
Fix #554 by exiting "gracefully" and giving more useful hint on why it failed. With this patch:

```
ERROR: Input path exceeds maximum supported length (255)
syntax:  acsccd [--help] [-t] [-v] [-q] [--version] [--gitinfo] input output
```

Out of scope: I have no WBS to charge to, so I cannot apply this to CALACS globally. This PR is only meant to solve the immediate problem reported originally over at ACSTOOLS.
closed 2021-07-30 by pllim