spacetelescope/hstcal #575: Remove calwf3 first-file dependency when processing members of an association table

starivera, 2022-10-20:
Currently when processing an association, calwf3 retrieves calibration switch and reference file keyword settings from the first image listed in the asn table and applies it to all images in the table. We need each individual exposure to use their own assigned reference files as specified in their header keywords during calibration.

In the past, stacked images were assumed to be taken from the same visit, which implied the same reference files being assigned and applied. However, in WFC3/UVIS post-flash is applied to mitigate the effects of CTE. The post-flash reference file depends upon the shutter blade used, which alternates from exposure to exposure (A-B-A-B). The post-flash reference file for shutter A and B are different, so applying one to an image taken with the other blade is undesirable. In addition, many WFC3 (and ACS) calibration images are taken throughout the year and reference files are updated periodically (e.g. BPIXTAB, DRKCFILE). Images from different epochs drizzled together will also have the wrong reference files applied if only the first file keywords are used.

mdlpstsci, 2022-10-28:
All four ASN datasets Sylvia provided were processed with CALWF3 from the latest HSTDP Release caldp_20221010.  Specifically, these datasets were analyzed to ensure the FLSHFILE used for each image in the association was, in fact, the FLSHFILE as specified in the primary header of the RAW file.  As Sylvia has noted, it is quite important the appropriate FLSHFILE is used for processing of the data.  In fact, it was explicitly demonstrated the FLSHFILE as documented in the header of the RAW image was the file used to process the data.
 
The major components of CALWF3 use the files and keyword values specified in the primary header of the RAW file being processed, _**EXCEPT for the cosmic ray rejection step, WF3REJ**_.  When preparing to perform the cosmic ray rejection, there are a few keywords which are read only from the first image.  These keywords are: DETECTOR, READNOISE, ATODGN, and CRREJTAB.  In the case of CRREJTAB, it is only read from the first image in the ASN list if it were not specified on the command line. (Aside: You may recall that to fix the bug which will allow you to specify the CRREJTAB on the command line, new releases of HSTCAL and WFC3TOOLS have to be released in a coordinated way.) Reading values from only the first image was implemented in 1999 for efficiency.
 
This email addresses only Sylvia’s concern regarding the application of calibration reference files corresponding to RAW images in an association.  I hope this email addresses and corrects my earlier misstatement and allays any concern about the use of the FLSHFILE.

In essence this ticket is the result of a misunderstanding.

mdlpstsci, 2022-10-28:
This ticket is associated with Jira HLA-870.

closed 2022-10-28 by mdlpstsci