spacetelescope/hstcal #588: Add support for WFC3/IR time-dependent IMPHTTAB to calwf3

astromariarosa, 2023-08-16:
`calwf3` was updated in 2020 to enable time-dependent photometry for the WFC3/UVIS detector (i.e. [issue #451](https://github.com/spacetelescope/hstcal/issues/451)). We would like to enact the same changes for the WFC3/IR detector by fall 2023, hopefully end of September. These are high-priority changes, is that timeframe possible? We can provide a dummy reference file for testing within the next two weeks.

mdlpstsci, 2023-08-16:
@astromariarosa I know about this work from some email discussion with Annalisa.  As such, this work is on my official schedule, but it is unlikely to be done by the end of September 2023.  I have some other high priority work which I need to address first, particularly since I did not expect this new IR calibration reference file until the September/October timeframe.

mdlpstsci, 2023-08-16:
@astromariarosa To be clear, there is ACS work which has higher priority than this work.  The WFC3 IR TVB work still has not been defined or passed to me for any action.

acalamid, 2023-08-16:
@mdlpstsci No problem. We still have to finalize the sensitivity slopes and re-calculate the inverse sensitivity values and  we won't have these until the end of September.
I was wondering if you managed to do some preliminary testing related to the ticket HLA-1030.
Thanks,
Annalisa

astromariarosa, 2023-08-17:
We have a dummy IMPHTTAB file ready for you, for whenever you get the time to work on this. It is located at `/grp/hst/wfc3v/marinelli/ir_phot/imphttab/ir_imphttab_td.fits`.

mdlpstsci, 2024-03-31:
@acalamid @astromariarosa 
FYI - I am taking a look at your dummy IMPHTTAB file.

mdlpstsci, 2024-04-16:
@acalamid @astromariarosa 
I have made the changes to calwf3 and compiled the code on the Linux systems for your testing.  You can access the modified calwf3 executable as: 

> /home/mdelapena/imphot_repos/hstcal/_build/pkg/wfc3/calwf3.e

You can test that you have the updated version:

> /home/mdelapena/imphot_repos/hstcal/_build/pkg/wfc3/calwf3.e --version 

and you should see 3.7.2.  If have any problems accessing this executable, please let me know.  Of course, if you have any issues/complaints/questions,  you should let me know too!

When comparing the FLT file as produced by the new version (3.7.2) to an older version, you will see:

```
(rc5_caldp_infrastructure) bash-4.2$ fitsdiff NEWOUT/*flt.fits OLDOUT/
 fitsdiff: 6.0.0
 a: /home/mdelapena/WFC3/IMPHTTAB/NEWOUT/ib4606clq_flt.fits
 b: /home/mdelapena/WFC3/IMPHTTAB/OLDOUT/ib4606clq_flt.fits
 Maximum number of different data values to be reported: 10
 Relative tolerance: 0.0, Absolute tolerance: 0.0
Primary HDU:
   Headers contain differences:
     Keyword CAL_VER  has different values:
        a> 3.7.2 (Apr-15-2024)
        b> 3.7.1 (Oct-18-2023)
     Keyword HISTORY [28] has different values:
        a>   reference table ir_imphttab_td.fits
        b>   reference table iref$4af1533ai_imp.fits
     Keyword HISTORY [30] has different values:
        a>     2023 Time-dependent Inverse Sensitivity
        b>     photometry keywords reference file---------------------------------
     Keyword IMPHTTAB has different values:
        a> ir_imphttab_td.fits
        b> iref$4af1533ai_imp.fits
     Keyword PHOTFLAM has different values:
        a> 1.5571101e-20
        b> 1.5318e-20
     Keyword PHOTFNU  has different values:
        a> 6.9102363e-08
        b> 6.7979137e-08
     Keyword PHOTMODE has different values:
        a> WFC3 IR F110W MJD#55611.0931
        b> WFC3 IR F110W

```

mdlpstsci, 2024-04-19:
Git PR#612 was merged onto main (HLA-1036).  Any changes requested by WFC3 team will be charged against a new JIRA HLA ticket.

mdlpstsci, 2024-05-06:
For my reference...
(Mac)
My test directory is /Users/mdelapena/HSTCAL/IMPHTTAB/IR.
Git branch is ir_imphot (repos/hstcal)
(Linux)
My test directory is /home/mdelapena/WFC3/IMPHTTAB.
Git branch is linux_ir_imphot (imphot_repos/hstcal/)

mdlpstsci, 2024-05-15:
@starivera 
Adding Isabel so she can follow-up on this issue.

astromariarosa, 2024-05-15:
@mdlpstsci Thanks for the nudge Michele! 

@st-apidgeon will be testing the updated executable. We will keep you in the loop regarding our progress and any issues we encounter.

mdlpstsci, 2024-06-04:
@st-apidgeon 
Please access the new executable as described on 16 April in this issue.  The new code is NOT on the main branch of HSTCAL.

st-apidgeon, 2024-06-05:
Hi @mdlpstsci, I was able to test the new executable using a dummy IMPHTTAB file on cluster data, and we saw the expected results with a ~1% change per year to PHOTFLAM in the outputs. We're still working on the final time-dependent IMPHTTAB and we can update you when that is ready.

mdlpstsci, 2024-06-05:
@st-apidgeon 
I am glad the software is working as you expect.  Having the "final" IR time-dependent IMPHTTAB for my testing will ensure there are no unexpected gotchas.

astromariarosa, 2024-08-02:
Hi @mdlpstsci, the new time-dependent IMPHTTAB file is located at 
`/grp/hst/wfc3v/marinelli/ir_phot/imphttab/ir_imphttab_2024-07-19.fits` 

Permissions should be fully open, please let me know if there are any issues.

mdlpstsci, 2024-08-06:
@astromariarosa FYI - Just confirming I was able to copy the latest version of the calibration file from the location specified.

mdlpstsci, 2024-09-03:
@astromariarosa 
The updated IR WFC3 time-dependent IMPHTTAB file checked out just fine.  Since @st-apidgeon checked out the software on 05 June successfully, and the GIT PR #621  has been approved by SCSB, I consider this Issue done.  This software will be part of the next science build which is anticipated mid-October 2024.

closed 2024-09-03 by mdlpstsci