spacetelescope/hstcal #611: error when running regression tests on GitHub Actions

zacharyburnett, 2024-04-02:
when running `pytest`:
```
pytest \
  --basetemp=/runner/_work/_temp/pytest_basetemp/  \
  --cov --cov-report=xml:/runner/_work/RegressionTests/RegressionTests/regtest/coverage.xml --cov-report term-missing \
  --override-ini junit_logging=system-err --junit-xml=/runner/_work/_temp/regtest/results.xml \
  -n auto --dist=loadscope \
  --color=no \
  --md-report --md-report-zeros empty --md-report-color never --md-report-flavor gfm --md-report-output /runner/_work/_temp/regtest/md_results.md \
  --bigdata --slow 
```

https://github.com/spacetelescope/RegressionTests/actions/runs/8524833382/job/23350417332#step:26:45
```
==================================== ERRORS ====================================
____________________ ERROR collecting tests/acs/test_hrc.py ____________________
tests/acs/test_hrc.py:6: in <module>
    from ..helpers import BaseACS
tests/helpers.py:53: in <module>
    set_exe_marker('acs')
tests/helpers.py:45: in set_exe_marker
    if os.path.isfile(cal):
<frozen genericpath>:30: in isfile
    ???
E   TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType
```
environment variables:
```
env:
    PYTHONUNBUFFERED: 1
    CI: false
    CRDS_PATH: /runner/_work/_temp/crds
    CRDS_SERVER_URL: https://hst-crds.stsci.edu/
    CRDS_CONTEXT: hst_1148.pmap
    TEST_BIGDATA: https://bytesalad.stsci.edu/artifactory
    MAMBA_ROOT_PREFIX: /home/runner/micromamba
    MAMBA_EXE: /home/runner/micromamba-bin/micromamba
    CONDARC: /runner/_work/_temp/setup-micromamba/.condarc
    PATH: /RegressionTests/bin:/home/runner/micromamba/envs/test-hstcal/bin:/home/runner/micromamba/condabin:/runner/_work/_temp/setup-micromamba:/home/runner/micromamba-bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/runner/.local/bin/
    jref: /grp/hst/cdbs/jref/
    iref: /grp/hst/cdbs/iref/
    oref: /grp/hst/cdbs/oref/
```

Am I setting up the environment correctly?

mdlpstsci, 2024-04-02:
@zacharyburnett I am not sure if this is your problem, but you also need ci-watson in your environment.

zacharyburnett, 2024-04-02:
I just noticed there are some passing tests on another branch; I'll take a look at those first to see what the differences are. Sorry for the false issue!

closed 2024-04-02 by zacharyburnett