spacetelescope/hstcal #631: TST: macos-12 job failed

pllim, 2025-03-17:
Example log: https://github.com/spacetelescope/hstcal/actions/runs/13773057638

pllim, 2025-03-18:
Two problems:

1. Replace macos-12 with macos-13.
2. But even after that, it still fails. Turns out Ubuntu is pulling in cfitsio 4.3.1 (passed) but brew pulling in cfitsio 4.6.0 (failed). OSX complains about unable to find `<fitsio.h>`.

I cannot figure out how to get brew to install older version of cfitsio (is that even possible)? Even if we can, should we look into why it is failing with cfitsio 4.6.0? It is quite a big jump from 4.3 to 4.6, so I think that is a bigger project that would require more extensive INS testing and so on.

pllim, 2025-03-18:
https://heasarc.gsfc.nasa.gov/fitsio/

* Mar 2025: CFITSIO Version 4.6.0 -- Released with vulnerability fixes along with other bug fixes and enhancements. (see [what's new](https://heasarc.gsfc.nasa.gov/FTP/software/fitsio/c/docs/changes.txt))
* Aug 2024: CFITSIO Version 4.5.0 -- Conversion of configure/build for automake. Bug fixes and enhancements.
* Jun 2024: CFITSIO Version 4.4.1 -- License update only, no code changes.
* Feb 2024: CFITSIO Version 4.4.0 -- released with bug fixes, enhancements and a reorganization to include additional FITS utility programs.

jhunkeler, 2025-03-18:
This is a bug on cfitsio's end.

Neither the path to the headers or the libraries are present in the installed `cfitsio.pc` file, so `pkg-config` returns a malformed argument list:

```
$ pkg-config --cflags cfitsio
-I
$ pkg-config --libs cfitsio
-L -lcfitsio
```

The expected output is:

```
$ pkg-config --cflags cfitsio
-I/path/to/include
$ pkg-config --libs cfitsio
-L/path/to/lib -lcfitsio
```

The configuration generated by 4.6.0's build system is:
```
prefix=/path/to
exec_prefix=/path/to
libdir=
includedir=

Name: cfitsio
Description: FITS File Subroutine Library
URL: https://heasarc.gsfc.nasa.gov/fitsio/
Version: 4.6.0
Libs: -L${libdir} -lcfitsio
Libs.private:  -lz -lcurl -lm
Cflags: -I${includedir}
```

Where `libdir` should have been set to `${prefix}/lib` and `includedir` should be `${prefix}/include`.

The bug in this case is that their `cfitsio.pc.cmake` template file contains this:

```
libdir=@LIB_DESTINATION@
includedir=@INCLUDE_INSTALL_DIR@
```

Those variables are not defined in `CMakeLists.txt` so they're rendered blank.

----

1. We can build `cfitsio` from source and patch it.
2. We can comment on https://github.com/HEASARC/cfitsio/issues/38 with a "Me too."
3. We can open a PR against `cfitsio` that replaces the undefined template variables with `libdir=${exec_prefix}/lib` and `includedir=${prefix}/include`. Using `pkg-config`'s internal variable expansion is easier to maintain than injecting multiple cmake variables.

From `pkg-config(1) :: Metadata File Syntax`:
> Files have two kinds of line: keyword lines start with a keyword plus a colon, and variable definitions start with an alphanumeric string plus an equals sign. Keywords are defined in advance and have special meaning to pkg-config; variables do not, you can have any variables that you wish (however, users may expect to retrieve the usual directory name variables).
> 
> Note that variable references are written "${foo}"; you can escape literal "${" as "$${". 

```diff
--- a/cfitsio-cfitsio-4.6.0/CMakeLists.txt      2025-03-13 11:16:56
+++ b/cfitsio-cfitsio-4.6.0/CMakeLists.txt      2025-03-18 15:57:49
@@ -40,6 +40,7 @@ INCLUDE(CheckCSourceCompiles)
 INCLUDE(CheckFunctionExists)
 INCLUDE(CheckSymbolExists)
 INCLUDE(CheckCSourceCompiles)
+INCLUDE(GNUInstallDirs)

 #==============================================================================
 # Build options:
```

```diff
--- a/cfitsio-cfitsio-4.6.0/cfitsio.pc.cmake    2025-03-13 11:16:56
+++ b/cfitsio-cfitsio-4.6.0/cfitsio.pc.cmake    2025-03-18 15:58:02
@@ -1,7 +1,7 @@
 prefix=@CMAKE_INSTALL_PREFIX@
-exec_prefix=@CMAKE_INSTALL_PREFIX@
-libdir=@LIB_DESTINATION@
-includedir=@INCLUDE_INSTALL_DIR@
+exec_prefix=${prefix}
+libdir=${exec_prefix}/@CMAKE_INSTALL_LIBDIR@
+includedir=${prefix}/include

 Name: cfitsio
 Description: FITS File Subroutine Library
```

4. We wait for the `cfitsio` maintainers to fix it.

pllim, 2025-03-18:
@jhunkeler , if you already have the diff, why not just propose the patch and claim victory points? Otherwise waiting it out is okay too.

pllim, 2025-03-18:
p.s. Thanks for the details! 💪

jhunkeler, 2025-03-19:
Done. It was slightly more involved than the diffs I posted above.

https://github.com/HEASARC/cfitsio/pull/39

pllim, 2025-04-01:
Looks like CFITSIO 4.6.2 is out. 🤞

pllim, 2025-04-01:
Passed now. Yay. Thanks, Joe!

closed 2025-04-01 by pllim