spacetelescope/hstcal #668: RFC: Move bias shift code to its own file

pllim, 2025-07-02:
@mdlpstsci said

https://github.com/spacetelescope/hstcal/blob/85c4cf2b6ac7cf94d03bd53c0b27370b31996b27/pkg/acs/lib/acsccd/doccd.c#L66

and

https://github.com/spacetelescope/hstcal/blob/85c4cf2b6ac7cf94d03bd53c0b27370b31996b27/pkg/acs/lib/acsccd/doccd.c#L367

should be moved out to its own file, so `doccd.c` can be clean like this

https://github.com/spacetelescope/hstcal/blob/85c4cf2b6ac7cf94d03bd53c0b27370b31996b27/pkg/acs/lib/acsccd/doccd.c#L745-L749

pllim, 2025-07-03:
@mdlpstsci , before I can proceed, I need to understand the current layout, so hope you can advise after you return from the holiday weekend:

* Why is DARKTIME computation in `doccd.c`? I thought DARKCORR happens in ACS2D step, not ACSCCD. I ask because it also uses the `subApertures` stuff you wanted me to move out of the file.
* As for the bias shift code you want me to move, is it cleaner if we move it to one of the existing C files instead of creating a new one? Which one? As it is, there is a bunch already and things are getting confusing.
* If you insist on a new C file, any suggestion on what to name it?

```
$ ls -1 pkg/acs/lib/acsccd/
acsccd.c
blevdrift.c
blevfit.c
blev_funcs_postsm4.c
doatod.c
dobias.c
doblev.c
doccd.c
dofwsat.c
dosink.c
findblev.c
findover.c
getacsflag.c
getccdsw.c
```

Thanks!

Ref: https://hst-docs.stsci.edu/acsdhb/chapter-3-acs-calibration-pipeline/3-3-structure-of-calacs

mdlpstsci, 2025-07-07:
@pllim 
In the History (Version 10.3.3) I see "Code written for CALACS 10.2.2 was moved from ACS2D to ACSCCD so every BLV_TMP file has the correct DARKTIME keyword value."

I see the header comment in doccd.c:
24-May-2021 MDD: Read the post-flashed and unflashed columns from the CCDTAB
     reference file to use for the offset to the DARKTIME FITS keyword value
     under appropropriate date and supported subarray criteria.  The DARKTIME
     keyword value is now the default scaling factor for the superdarks, with
     the offset being an additive correction to DARKTIME under appropriate
     circumstances. The offset values is applicable for WFC and HRC only.
     The code was moved from acs2d/dodark.c which contained the original
     implementation.

This was done because Issue#490, PR#551.

closed 2025-07-24 by pllim