spacetelescope/hstcal #669: Why are we using raw assert?

pllim, 2025-07-03:
https://github.com/search?q=repo%3Aspacetelescope%2Fhstcal%20assert&type=code

Example:

https://github.com/spacetelescope/hstcal/blob/85c4cf2b6ac7cf94d03bd53c0b27370b31996b27/pkg/acs/lib/acsccd/blev_funcs_postsm4.c#L61-L64

Shouldn't we log the proper error code and exit gracefully like this?

https://github.com/spacetelescope/hstcal/blob/85c4cf2b6ac7cf94d03bd53c0b27370b31996b27/pkg/acs/lib/acsccd/blev_funcs_postsm4.c#L84

We don't want the pipeline to panic and abort, right?

Ref: https://en.cppreference.com/w/cpp/error/assert.html

mdlpstsci, 2025-07-07:
@pllim 
You are quite correct in that a raw _assert_ should **not** be used in any pipeline code!  Any bug/failure/exception should be handled with an informative message and a graceful exit.

pllim, 2025-07-07:
The fact the pipeline never panics here (I think we would have heard from DMS if it did) probably means this check is unnecessary. We do high level checks further up in `doxxx.c`, no? And some logic is set by flags that are out of our control. So basically by the time you get here, you get what you ask for (i.e., garbage in, garbage out). So here are 2 options:

1. Remove the checks entirely but leave the comment for future with implicit assumptions.
2. Change the checks to proper `trlerror` stuff, though this might duplicate checks already made elsewhere higher up the chain.

Do you have a preference, @mdlpstsci ?

mdlpstsci, 2025-07-07:
@pllim 
In truth these checks definitely seem redundant.  If the processing has gotten this far, the checks seem entirely unnecessary.  I would consider just removing them entirely.